tsduck/tsduck #465: Question about modifying ts header

meguli, 2019-11-21:
I want to use _tsduck_ to test my application which listens for transport stream on an IP address and answers some questions. Suppose my program listens on `239.48.208.1:1234`. 

I can use _tsduck_ plugins to introduce `ts_stream_error`, play with pcr settings etc. and I am able to see on my program that modifications are applied to the output stream on `239.48.208.1:1234`. One thing I want to add to my program is to count sync byte errors and sync losses. For that, I want to output ts packets with sync byte that are not equal to `0x47`. It looks to me like I am not able to do this with _tsduck_. Here is one of the things that I tried

    tsp -I file --infinite "playout.ts" -O ip 239.48.208.1:123 --ttl 10

`playout.ts` contains a ts packet with sync byte set to  `0x77`. And I get the error

> Error: file: synchronization lost after 0 packets, got 0x77 instead of 0x47

immediately, which makes sense. Still, this is exactly what I want to playout. Does anyone know a way to output a stream with incorrect sync byte?

lelegard, 2019-11-21:
>  Does anyone know a way to output a stream with incorrect sync byte?

Most tools will probably refuse to do so since they synchronize themselves on 0x47. Furthermore, how is transmitted a TS packet? Modulators will probably reject invalid packets. And if they don't, the demodulator will reject the packets.

With UDP/IP, what does your application expect? Does it look for raw TS packets in UDP packets, RTP headers, other encapsulation? Or, does it (as `tsp -I ip` does), look for a sequence a 0x47 bytes at regular distances of 188 bytes to identify the TS packets regardless of the encapsulation?

Probably, the first question to ask is : what do you want to test ?

meguli, 2019-11-21:
I try to look for raw TS packets in UDP packets as you first pointed out. The idea is to analyze incoming TS packets for various errors and statuses. And in this particular case, I want to detect and see things like; if there are synchronization errors, what is the value of sync byte that caused the sync to be lost etc.

It makes sense for _tsduck_ to not output the stream in this case so it is quite possible that _tsduck_ is not the tool to use for this job. But I was hoping maybe it is possible. Something like;

> tsp -I file --infinite --ignore-synchronization-problems "playout.ts" -O ip 239.48.208.1:123 --ttl 10

or maybe an output plugin to introduce sync byte errors every `N` packets.

lelegard, 2019-11-21:
So, if this is just for a test, you do not need a specialized tool such as TSDuck.

On Linux, you can do that using the shell only. Assuming, that you have a large TS file with proper byte corruption, first split your TS file in chunks of 7 TS packets (the usual number of TS packets per UDP packet);
~~~
$ split --bytes $((7*188)) input.ts foo- --suffix-length 4 --additional-suffix .ts
$ ls -l foo-*
-rw-r--r-- 1 user group   1316   foo-aaaa.ts
-rw-r--r-- 1 user group   1316   foo-aaab.ts
-rw-r--r-- 1 user group   1316   foo-aaac.ts
-rw-r--r-- 1 user group   1316   foo-aaad.ts
-rw-r--r-- 1 user group   1316   foo-aaae.ts
-rw-r--r-- 1 user group   1316   foo-aaaf.ts
~~~
Then, send them one by one, waiting one millisecond between each:
~~~
for f in foo-*.ts
do
    socat -u open:$f udp:localhost:1234
    sleep 0.001
done
~~~
Just tested on an Ubuntu 19.10 machine. It works (received using `tsp` on the other side).

meguli, 2019-11-21:
After doing what you said, I am also able to playout on `localhost:1234` and verify on _tsduck_ using

> tsp -I ip -s 127.0.0.1 1234 -O ip 239.48.208.1:1234 --ttl 10

works all fine and I am able to analyze the stream output of _tsduck_ on  address _239.48.208.1:123_ using my application. Everything is working fine with one caveat of course, _tsduck_ is dropping packets with incorrect sync byte. This is expected of course. So i tried multicasting directly to the address instead using something similar to what you gave.

    socat -u open:sync_byte_broken.ts udp:239.48.208.1:1234
    sleep 0.001

This time, I am not receiving anything on that address at my application. (unlike above _tsduck_ relay). So clearly, that comment is not how you multicast something to an address using _socat_. Seemed like could just work. Anyways, I guess this concludes the _tsduck_ related part of this question though. Thanks for the help.

closed 2019-11-22 by meguli