tsduck/tsduck #466: Latest version not opening dvb

joolzg, 2019-11-21:
fred@home:~$ tslsdvb 
tslsdvb: error getting delivery systems of /dev/dvb/adapter0/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter1/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter2/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter3/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter4/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter5/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter6/frontend0: Invalid argument
tslsdvb: error getting delivery systems of /dev/dvb/adapter7/frontend0: Invalid argument

fred@home:~$ tsp -v -I dvb --adapter 7 --frequency 12253000000 --symbol-rate 20000000 --polarity vertical   --lnb 9750,11250,11700 -P until --seconds 10 -P analyze -o out_12253.txt -P eit -o eit_12253.txt -O file a.ts
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* eit: creating eit_12253.txt
* Error: dvb: error getting delivery systems of /dev/dvb/adapter7/frontend0: Invalid argument
* dvb: using  (none)
* Error: dvb: tuner not open
* Error: dvb: DVB tuner not open

No when i built dvblast i had to change the config to downgrade the DVBAPI_VERSION, i used 450 as a value and that worked.

Any help as I want to test the ATSC stuff

joolzg, 2019-11-21:
OK found the same option and rebuilt with 450 as the version number, but it still fails.

joolzg, 2019-11-22:
So today I tried to bisect to find the problem and after lots of failed test and failed builds i ended up here
So on master, 085d14e82002272a0da3d4a6b9642e6e56a62c73

git bisect bad

git checkout e10daf28eaa46e0728a27e9c716ef93f1d910ad2

git bisect good
Bisecting: 1 revision left to test after this (roughly 1 step)
[168b1386ff280869acfbfafde10632c66fe66de5] More tuner args
[CXX] tsDVBCSA2.cpp
In file included from linux/tsDTVProperties.h:158:0,
                 from linux/tsDTVProperties.cpp:34:
linux/tsDTVPropertiesTemplate.h:37:65: error: expected ',' or '...' before '&' token
 void ts::DTVProperties::getValuesByCommand(std::set<INT>& values& values, uint32_t cmd) const

git bisect bad
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[b319353246e655f14ba839c440c03e10a1f2f3d6] Started tuner args refactoring
[CXX] tsDVBCharset.cpp
In file included from linux/tsDTVProperties.h:158:0,
                 from linux/tsDTVProperties.cpp:34:
linux/tsDTVPropertiesTemplate.h:37:65: error: expected ',' or '...' before '&' token
 void ts::DTVProperties::getValuesByCommand(std::set<INT>& values& values, uint32_t cmd) const

git bisect bad
b319353246e655f14ba839c440c03e10a1f2f3d6 is the first bad commit
commit b319353246e655f14ba839c440c03e10a1f2f3d6
Author: Thierry Lelegard <thierry@lelegard.fr>
Date:   Mon Oct 7 22:13:18 2019 +0200

    Started tuner args refactoring

:040000 040000 9751245e2e6f6c3656cb7d1153ba81a25efbffbb 72bcaea9cb89bd15d498674df08bce6aa5fabceb M	build
:040000 040000 100d36d1c2bec2310b6ce589858879dcde8c7384 de4eaa7b43e644b33d4a15460a2c6bb61adfb9d1 M	src

I hope this helps as its seems to have broken in the past few commits

lelegard, 2019-11-22:
Hi @joolzg 

> i had to change the config to downgrade the DVBAPI_VERSION, i used 450 as a value and that worked.

What do you mean by that? Which distro and kernel version are you using?

I use kernel versions from 3.10 to 5.3 on Fedora, CentOS, Ubuntu and the latest versions of TSDuck work with tuners. I do not have ATSC tuners (and no ATSC network) but the failing command that you mention clearly uses a DVB-S tuner. My DVB-S tuner works.

I wonder what you mean by _"downgrade the DVBAPI_VERSION, i used 450"_. The oldest DVB API I can see is 5.10, in kernel 3.10. In kernel 5.3, the DVB API version is 5.11. So, this API has been relatively stable for years.

The DVB API switched from V4 to V5 about 10 years ago, after the "S2 API great war" that I remember quite well. I wonder how you can have DVB API v4 on recent kernels.

Anyway, in the meantime, the DVB support has greatly improved in the kernel, especially supporting multi-standard tuners. The TSDuck code about DVB tuner was rather old and did not take advantage of this. Not supporting multi-standard tuners was a problem that I recently solved using improvements in the DVB API. To do that, I had to make a lot of modifications, including the commit you identified. I cannot support at the same time very old kernels (10+ years old) and recent ones with the same software.

joolzg, 2019-11-22:
What I mean is to get dvblast working i had to modify config.h to set DVB_API_VERSION to below 500

As pointed out with your new code it does not recongnise the dvb card anymore.
as an example
fred@home:~$ dvblast -a 7 -f 1111
DVBlast 2.2 (git-2.2-27-g51c9565)
warning: restarting
debug: compiled with DVB API version 5.4
debug: using DVB API version 3.0
debug: Frontend "TurboSight TBS 6904 DVBS/S2 frontend" supports:
debug:  frequency min: 950000, max: 2150000, stepsize: 1011, tolerance: 5000
debug:  symbolrate min: 1000000, max: 45000000, tolerance: 0
debug:  capabilities:
debug:   INVERSION_AUTO
debug:   FEC_1_2
debug:   FEC_2_3
debug:   FEC_3_4
debug:   FEC_4_5
debug:   FEC_5_6
debug:   FEC_6_7
debug:   FEC_7_8
debug:   FEC_AUTO
debug:   QPSK
debug:   2G_MODULATION
debug:   FE_CAN_RECOVER
debug:  delivery systems:
debug:   DVBS
debug:   DVBS2

tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.19-1447
Works

Master gives the above errors

uname -a
Linux joemex6 4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

tbs driver in use
tbs-linux-drivers_v170330.zip

Do you need anymore information, as this is not specific to ATSC its not identifying the card now.

BR

Joolz

lelegard, 2019-11-22:
So, what you mean is that you modified a config file in dvblast source code, right? In that case, it has nothing to do with the kernel nor TSDuck. So, TSDuck should compile and behave the same way as I see it on my kernel. Is there any specific config on your system?

That being said, if dvblast does not not work with DVB API more recent than 10 years old, it becomes "abandonware". You should be prepared to either drop it or  maintain it yourself.

joolzg, 2019-11-22:
It has something to do with TBS drivers, but as pointed out even doing the same modification to tsPlatform.h to use the same value, 450, makes no difference tsduck still fails to recognize the card.

And it was working until around the 10th of October.

Its a shame as I have been looking for a tool that does ATSC well, and the changes you made were helping a lot. 

BR

joolz

joolzg, 2019-11-22:
Looking at the code in tsTunerGuts.cpp
    if (::ioctl(_guts->frontend_fd, ioctl_request_t(FE_GET_INFO), &_guts->fe$
        report.error(u"error getting info on %s: %s", {_guts->frontend_name,$
        return close(report) || false;
    }
    _guts->fe_info.name[sizeof(_guts->fe_info.name) - 1] = 0;
    _device_info = UString::FromUTF8(_guts->fe_info.name);

    // Get the set of delivery systems for this frontend.

    _delivery_systems.clear();
    DTVProperties props;
    props.add(DTV_ENUM_DELSYS);
    if (::ioctl(_guts->frontend_fd, ioctl_request_t(FE_GET_PROPERTY), props.$
        report.error(u"error getting delivery systems of %s: %s", {_guts->fr$
        return close(report) || false;
    }
    props.getValuesByCommand(_delivery_systems, DTV_ENUM_DELSYS);

But in dvblast, dvb.c
    if ( ioctl( i_frontend, FE_GET_INFO, &info ) < 0 )
    {
        msg_Err( NULL, "FE_GET_INFO failed (%s)", strerror(errno) );
        exit(1);
    }

    uint32_t version = 0x300;
#if DVBAPI_VERSION >= 505
    if ( ioctl( i_frontend, FE_GET_PROPERTY, &info_cmdseq ) < 0 )
    {
#endif
        /* DVBv3 device */
        switch ( info.type )

joolzg, 2019-11-22:
dvblast, dvb.c has this code
    if ( ioctl( i_frontend, FE_GET_INFO, &info ) < 0 )
    {
        msg_Err( NULL, "FE_GET_INFO failed (%s)", strerror(errno) );
        exit(1);
    }

    uint32_t version = 0x300;
#if DVBAPI_VERSION >= 505
    if ( ioctl( i_frontend, FE_GET_PROPERTY, &info_cmdseq ) < 0 )
    {
#endif
        /* DVBv3 device */
        switch ( info.type )
        {
        case FE_OFDM:
            p_systems[i_systems++] = SYS_DVBT;

tsduck, tsTunerGuts.cpp
    if (::ioctl(_guts->frontend_fd, ioctl_request_t(FE_GET_INFO), &_guts->fe_info) < 0) {
        report.error(u"error getting info on %s: %s", {_guts->frontend_name, ErrorCodeMessage()});
        return close(report) || false;
    }
    _guts->fe_info.name[sizeof(_guts->fe_info.name) - 1] = 0;
    _device_info = UString::FromUTF8(_guts->fe_info.name);

    // Get the set of delivery systems for this frontend.

    _delivery_systems.clear();
    DTVProperties props;
    props.add(DTV_ENUM_DELSYS);
    if (::ioctl(_guts->frontend_fd, ioctl_request_t(FE_GET_PROPERTY), props.getIoctlParam()) < 0) {
        report.error(u"error getting delivery systems of %s: %s", {_guts->frontend_name, ErrorCodeMessage()});
        return close(report) || false;
    }
    props.getValuesByCommand(_delivery_systems, DTV_ENUM_DELSYS);

If it is at all possible for you to modify the code to allow this older DVB_API as dvblast does it would mean i can take advantage of your future ATSC modifications.

BR

Joolz

lelegard, 2019-11-22:
That is the whole point. The command `DTV_ENUM_DELSYS` is here to enumerate the "delivery systems" of the tuner.

In the old API, the `FE_GET_INFO` command which returns various information on the tuner also returned one single delivery system, e.g. DVB-T, ATSC, etc. So, when a tuner supported more than one system, only one could be used.

Look at the code in `/usr/include/linux/dvb/frontend.h` (on a rather old kernel 3.10 from 2013):
~~~
struct dvb_frontend_info {
    char name[128];
    fe_type_t  type;  /* DEPRECATED. Use DTV_ENUM_DELSYS instead */
~~~

The `info.type` which is used by dvblast has been deprecated for years. And it prevents the usage of multi-standard tuners. So, it is more than time to move.

In your case, your TBS driver must be very old if it does not support the `DTV_ENUM_DELSYS` command, much older than the kernel you are running it in. A driver is something that is very dependent on the kernel. You need to find a more recent driver. If the driver is old and so is the device, the stock kernel may include it. Does it?

I cannot make TSDuck compatible with many old versions of DVB API. Around 10 years ago, when the S2 API (aka v5) was introduced, I made a lot of modifications in the code to support both, I mean dynamically to take into account various drivers. The code was awful and error prone. Years after, I cleaned it up to drop v4 support and it was a relief. I have no plan to redo the same thing to support drivers which do not support an 8 years old command.

joolzg, 2019-11-22:
Lastest working drivers for TBS that work so I will have to live with the last working tsduck.

joolzg, 2019-11-22:
FYI, i added ~20 lines of code from the working tsduck at the point where FE_DELSYS fails, to use the fe_info type. 

For my case where the card is DVB-S/S2 it gives the correct values and now works.

So I will keep this in my local stash and when you add all the great features i can still use them.

BR

Joolz
ps. Thanks for this great software, only wish other people **TBS** could produce something as polished

lelegard, 2019-11-22:
> FYI, i added ~20 lines of code from the working tsduck at the point where FE_GET_PROPERTY fails, to use the fe_info type.

OK, I just pushed a similar workaround. Could you please test it?

Too bad I just released version 3.19 😞

joolzg, 2019-11-23:
Nearly the same code as mine, apart from I have the extra parts to add DVB-S2, T2 and C2 as with this modifcation, and mine, no changes to the DVB_API_VERSION are needed.

joolzg, 2019-11-23:
Tested your version and it works, albeit that it only adds DVB-S and not DVB-S2, see comment above

lelegard, 2019-11-23:
Added support for S2/T2/C2.

joolzg, 2019-11-23:
Thanks for the modifications, works well.

closed 2019-11-23 by joolzg