tsduck/tsduck #467: Suggestion: Dynamic pid filtering

lars18th, 2019-11-22:
Hi @lelegard ,

I want to share this idea about a new functionality. Please, consider it only if you feel it has sense.

The idea comes from tuners and it mimics the functionality of network tuners (SAT>IP or HDHR). I feel it can be implemented inside the current `filter` plugin or in a new one. And the functionality is based on the remote control of the `tsswitch`.

An example will easily illustrate it: `-P pid-filter --remote 127.0.0.1:4444`
The you can use:
```
echo >/dev/udp/127.0.0.1/4444 all
echo >/dev/udp/127.0.0.1/4444 0,1,16,17,20
echo >/dev/udp/127.0.0.1/4444 0,100,101,102
echo >/dev/udp/127.0.0.1/4444 none
```

I feel it's self explanatory.

One use case (you can found others): You are using a large FILE with a full captured MPTS. You want to use something like the `zap` plugin but dynamically. With this new functionality you can filter the stream on-demand.

What you think?

lelegard, 2019-12-01:
Hi @lars18th 

I have implemented this in a much more generic way. I have added a command named `tspcontrol` which can send control commands to a running `tsp`. One of these commands restarts any plugin with new parameters.

Example:
~~~
$ tsp --control-port 5555 \
      -I file -i streams/france-dttv/tnt-uhf22-482MHz-2019-01-22.ts \
      -P regulate \
      -P filter --pid 0-1 \
      -P tables \
      -O drop
~~~

From another session:
~~~
$ tspcontrol -t 5555 list -v

Executable: .../tsduck/src/tstools/release-x86_64/tsp

 0: -I file -i streams/france-dttv/tnt-uhf22-482MHz-2019-01-22.ts
 1: -P regulate
 2: -P filter --pid 0-1
 3: -P tables
 4: -O drop

$ tspcontrol -t 5555 restart 2 --pid 16-20
$ tspcontrol -t 5555 list -v

Executable: .../tsduck/src/tstools/release-x86_64/tsp

 0: -I file -i streams/france-dttv/tnt-uhf22-482MHz-2019-01-22.ts
 1: -P regulate
 2: -P filter --pid 16-20
 3: -P tables
 4: -O drop

$ tspcontrol -t 5555 exit
~~~

There are a couple of other control commands such as `suspend` and `resume` to temporarily bypass a plugin. See the user's guide as usual.

Note that I have not tested this on many plugins. So far, plugins have been only tested in one single scenario: start, process packets, close. Now, many such cycles are possible. It is possible that some plugins do not behave correctly if the initial state is not properly reset during a start _after_ a close. I had to fix a couple of things in the plugin `tables` for instance. If you find an incorrect plugin behaviour after a restart, please report it here.

lars18th, 2019-12-13:
Hi @lelegard ,

As usual, your implementation exceeds expectations.
Thank you!

closed 2020-01-13 by lelegard