tsduck/tsduck #468: Latest git not building

joolzg, 2019-11-25:
[CXX] tsArgsWithPlugins.cpp
In file included from tsArgsWithPlugins.h:36,
                 from tsArgsWithPlugins.cpp:34:
tsArgs.h:561:22: error: ‘virtual bool ts::Args::analyze(const ts::UString&, bool)’ was hidden [-Werror=overloaded-virtual]
         virtual bool analyze(const UString& command, bool processRedirections = true);
                      ^~~~~~~
In file included from tsArgsWithPlugins.cpp:34:
tsArgsWithPlugins.h:74:22: error:   by ‘ts::ArgsWithPlugins::analyze’ [-Werror=overloaded-virtual]
         virtual bool analyze(const UString& app_name, const UStringVector& arguments, bool processRedirections = true) override;
                      ^~~~~~~

lelegard, 2019-11-25:
I know. It builds with LLVM / clang on macOS (my primary development platform) but issues a fatal warning with GCC and MSC. Fixing it, stay tuned...

lelegard, 2019-11-25:
Fixed

closed 2019-11-25 by lelegard

joolzg, 2019-11-25:
Thanks just wanted you to know and i dont expect you to be online 24/7