tsduck/tsduck #469: Suggestion: timeref plugin, only update EIT date, not time

mcflyuk, 2019-11-25:
Hi @lelegard, I've been reading through all the TSDuck documentation and testing with a Dektec DTU-215, it's definitely a useful tool and great job putting it together. 

I'm particularly interested in the timeref plugin for modifying EIT within an existing transport stream. Using _-P timeref --start system --eit_ will modify the stream TDT to the current time and then start the EIT schedule from the current time as well.

However when you have a live broadcast schedule, this ends up with daytime TV events in the middle of the night. In the UK where there is the concept of the watershed (i.e. content not meant for children shown after 9pm), this is an issue as well.

I'd still like to be able to call the timeref plugin to change to stream to match the current time, but to have only the EIT date updated to match the system date, not the EIT start times.

I've been looking through the TSDuck documentation to look for a solution but couldn't find anything, hence submitting this on Github. What I'm suggesting is effectively an _--eitdateonly_ parameter to the timeref plugin. I've pulled the source code to investigate, however I don't have a C++ background so I've not had any luck trying to modify this myself.

Do you think this functionality is possible to achieve in TSDuck?

lelegard, 2019-11-25:
> to have only the EIT date updated to match the system date, not the EIT start times.

What do you mean by "EIT date" vs. "EIT start time"?

mcflyuk, 2019-11-26:
Thanks for the quick reply. The events in the EIT are signalled with a 40-bit start_time field (DVB EN 300 468).

> start_time: This 40-bit field contains the start time of the event in Universal Time, Co-ordinated (UTC) and Modified Julian Date (MJD) (see annex C). This field is coded as 16 bits giving the 16 LSBs of MJD followed by 24 bits coded as 6 digits in 4-bit Binary Coded Decimal (BCD). If the start time is undefined (e.g. for an event in a NVOD reference service) all bits of the field are set to "1".
> EXAMPLE 2: 93/10/13 12:45:00 is coded as "0xC079124500".

The EIT date is the first 16 bits of the start_time field, the EIT start time is the last 24 bits of the start_time field.

I've not been able to locate it in the source code but as the date requires an MJD operation whereas the time is a BCD, my suspicion is that the code to perform these two operations are separate? And then merged together to update the start_time field in the EIT?

So by updating the first 16 bits of the start_time (EIT date) but leaving alone the last 24 bits (EIT start time), a transport stream could have its date changed but leave the start times to match the original broadcast schedule.

Does this help explain it better?

lelegard, 2019-11-26:
> Does this help explain it better?

OK, clear, this is the date and time fields of the same date/time. I thought you meant two different timestamps but since an event has only one start date/time, I did not understand where the second date/time came from.

> I've not been able to locate it in the source code 

It is serialized [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsEIT.cpp#L316) and deserialized [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsEIT.cpp#L211).

So, you would like to change the _date_ only in each event, leaving the _hour/min/sec_ unchanged, isn't it?

mcflyuk, 2019-11-27:
> So, you would like to change the _date_ only in each event, leaving the _hour/min/sec_ unchanged, isn't it?

That's exactly it :-)

Thanks for providing the code references, I had found tsEITProcessor.cpp but not yet seen tsEIT.cpp.

So DecodeMJD is getting the date/time values from the EIT (or other tables), and EncodeMJD is writing a new date/time? There's also DecodeBCD/EncodeBCD but that looks like it's only used for event duration.

It looks like the code starts by getting the offset in milliseconds [here](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_timeref.cpp#L284), and then setting it as the _start_time_offset variable. BTW, very minor typo that doesn't affect the code - addStartTimeOffet presumably should be addStartTimeOffset?

Then it gets the event time from [DecodeMJD](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsEITProcessor.cpp#L376), [applies the _start_time_offset](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsEITProcessor.cpp#L380) before writing the new time back to the event [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsEITProcessor.cpp#L381).

As the offset is all done in milliseconds, the code would need to take of copy of the time variable before applying the _start_time_offset. The [EncodeMJD](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsMJD.cpp#L94) function actually processes the [time](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsMJD.cpp#L102) as well as the [date](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsMJD.cpp#L110).

Perhaps one way to implement it would be to add a getDate and getTime function in tsMJD.cpp, which the EITProcessor::handleSection function could call - similar to DecodeMJD. Then have a flag in tsEITProcessor which if true, update the time variable but then overwrite the updated time with the original start time from the event. Alternatively, get the date either from _--start system_ or the date provided and write that to the time variable, without applying the time += _start_time_offset at all.

You may be able to tell I've been working this through in my head as I've typed this! Bit late to have a go at implementing the code myself tonight, you might have a better idea for how to go about it considering your experience with the code base? The problem is how to implement it in a clean and tidy way that is controlled from the tsplugin_timeref.cpp code.

lelegard, 2019-11-28:
Option `--eit-date-only` was added in commit 5f21b3eaab796bce94da0c9e00c4e087fa6d925b

This was an easy one. It is only this:
~~~
-            if (!EncodeMJD(time, data + 2, MJD_SIZE)) {
+            if (!EncodeMJD(time, data + 2, _date_only ? MJD_MIN_SIZE : MJD_SIZE)) {
~~~
All the rest is cosmetics only.

mcflyuk, 2019-11-28:
Wow, I hadn't considered implementing it in that way, that makes much more sense!

I've pulled the commit and re-built the executable, started doing some checks on streams using the new --eit-date-only parameter with --start system. I'll report back on my findings but early indications are positive.

lelegard, 2019-11-28:
Make to get the latest repo (`git pull`), not only this commit. I fixed another detail later.

mcflyuk, 2019-11-29:
Have taken the latest repo and rebuilt. Was wondering why I couldn't output to the Dektec modulator and was getting an error using _-O dektec_ . But then I downloaded the Dektec SDK and now the Build.ps1 command is pulling in the DTAPI files.

The new command is working great! 👍 Thanks for the added functionality, I have a further question but that's unrelated to the timeref plugin so I'll raise it as a separate query.

If you have a paypal or gofundme message me the info and I'll contribute to your beer fund to say thanks :-)

closed 2019-11-29 by mcflyuk