tsduck/tsduck #470: Buffer size in kilo bytes for better flow control

igorauad, 2019-11-25:
Hi @lelegard ,

Is there an option to specify `--buffer-size-mb` in kilo-bytes rather than mega-bytes?

I'm experimenting with a setup that relies on `leandvb` in order to decode DVB-S2 in real-time and pipe the resulting MPEG TS into `tsp -P mpe`. The pipeline looks like so:

```
rtl_sdr -g 40 -f 1100e6 -s 2e6 - | \
 ./leandvb -f 2e6 --sr 1000e3 --roll-off 0.2 --standard DVB-S2 --sampler rrc | \
 tsp --realtime -P mpe --pid 32-33 --udp-forward --local-address 127.0.0.1 -O drop
```

The default buffer size of 16 MB was too large for this. When using it, I was having very bursty UDP output and, as a result, Linux was dropping some of the UDP datagrams that should ultimately reach an application.

So I changed the buffer size to 1 MB and also included the `regulate` plugin:
```
rtl_sdr -g 40 -f 1100e6 -s 2e6 - | \
 ./leandvb -f 2e6 --sr 1000e3 --roll-off 0.2 --standard DVB-S2 --sampler rrc | \
 tsp --realtime --buffer-size-mb 1 -P regulate -b 1200000 -P mpe --pid 32-33 --udp-forward --local-address 127.0.0.1 -O drop
```

However, this didn't seem to be enough. Due to clock differences, I believe at some point the specified bitrate of `regulate` won't match with the source bitrate (the output of `leandvb`) and this will fail. Also, in the use case I'm working on, it would be disadvantageous to require specification of the output bit rate on the receive side. This would reduce the flexibility on rates on the Tx side, like for example with DVB-S2 ACM.

Instead, the solution that does seem to work well in my setup is to have a sufficiently short input buffer, in the order of kBytes.  With that, it seems that I don't even need `regulate`, since the flow is already inherently controlled by the real-time reception. I'm now running `tsp` with 1 kB input buffer, after changing [tspOptions.cpp#L245](https://github.com/tsduck/tsduck/blob/master/src/tstools/tspOptions.cpp#L245). I wonder if it would be interesting to add an option for specifying the input buffer size in kB rather than MB? Or maybe there is another way to control this? I tried playing with `--max-input-packets` and `--max-flushed-packets` as per the user guide, but with not much success.

To give you an idea about rates, I'm testing reception of a 1 MBaud DVB-S2 signal. The output MPEG TS stream is roughly around 1.2 Mbps

Thanks

lelegard, 2019-11-25:
Your problem is about latency and bursts. This is not exactly the same thing as the buffer size. It is true that, by default, `tsp` favors throughput and reliability (ie. not losing packets) over latency and bursts. The option `--realtime` improves the latency and reduces bursts by adjusting some parameters, but probably not enough for your situation. You need to tune those parameters a bit more.

Your solution is probably to reduce both `--max-input-packets` and  `--max-flushed-packets`. See the explanations in the user's manual.

Dramatically reducing the buffer size reduces the latency at the expense of an increased risk of packet loss. This is why I do not recommend it. I see no good reason to reduce the buffer size below 1MB. It is too risky and does not bring anything if other parameters are fine tuned.

lars18th, 2019-11-29:
Hi,

I only want to add my 5 cents:

> I see no good reason to reduce the buffer size below 1MB. It is too risky and does not bring anything if other parameters are fine tuned.

I'm sure you're right, and the problem in this case comes from jitter and burst errors. _However_, from the point of view of an engineer good tools are not too restrictive. If someone needs to use a library/tool in a way that is not designed, then it's their risk. Good engineers know the limits, and perhaps for some research they need to check some unusual values.

After I've said this, I make this little suggestion:

- To not change the interface, continue maintaining the MB scale for the `--buffer-size-mb` parameter. But it accepts decimal values like `--buffer-size-mb 0.1`.

I hope you agree.
Regads.

igorauad, 2019-12-01:
Hi @lelegard ,

Thanks for the information. I've played more with the `--max-flushed-packets` and `--max-input-packets` options and I was able to get a reasonably stable output bitrate with:

```
tsp --realtime --buffer-size-mb 1 --max-flushed-packets 10 --max-input-packets 10
```

Yet, I think the change proposed by @lars18th is quite reasonable for flexibility, i.e. to allow decimal values like `--buffer-size-mb 0.1`.

Regards

lelegard, 2019-12-02:
OK, decimal values are now allowed for `--buffer-size-mb`. It was the opportunity to add fixed-point values in command line arguments. This is needed for some future improvements.

But there is still no good reason to decrease the `tsp` buffer size below 1 MB after fine tuning other parameters.

igorauad, 2019-12-02:
@lelegard That's great. Thank you very much again.

closed 2019-12-02 by igorauad