tsduck/tsduck #471: -O RTP option timestamp

themoog, 2019-11-26:
Hi Chaps, 

I'm trying to generate a test MPEG-TS RTP stream with Null MPEGTS packets.

this is my command line

tsp -I craft -P regulate -b 1000000 -O ip -p 7 -e --local-address 172.16.0.178 -r 239.0.0.1:1234

Cant get my analyser to lock

Looking at the RTP capture the timestamps are all 0?

[image]

Is this something to do with PCR?

Any thoughts on how I can get this to work?

Neil

lelegard, 2019-11-26:
> Is this something to do with PCR?

Yes, definitely. The RTP timestamp is a network reference clock. And the source reference clock of a TS is the PCR. So, the RTP time is synchronized with the PCR. See the comments [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsIPOutputPlugin.cpp#L308).

> Any thoughts on how I can get this to work?

You need a reliable PCR in the TS.

themoog, 2020-12-05:
HI Chaps,

Picking this one back up again...

tsp -b 10000000 -I null -P regulate -b 10000000 -O ip --rtp -l 192.168.1.20 239.1.1.1:20000

Did get RTP timestamps going, but still no joy with my end device. 

Has anyone got any examples of generating a MPEG-TS test transport stream from scratch for testing? I want to try and do some automation 👍 

Thanks in advance

Neil

lelegard, 2020-12-05:
You should start by the question : what does your receiver device need to analyse the stream?

PCR? PSI/SI? Non-null packets? Once you know what you need, we can try to find a way to provide it using TSDuck.