tsduck/tsduck #473: rtsp in TSDUCK

engineer507, 2019-11-27:
Hello
how i can add rtsp streaming in tsduck?
is there any rtsp output plugin for tsduck?
thanks
robert

lelegard, 2019-11-27:
Hi @engineer507,

There is no RTSP plugin in the TSDuck code base, neither input nor output plugin. There is only RTP (option `--rtp` in `ip` output plugin, automatically detected - and removed - in `ip` input plugin).

I am not aware of any RTSP implementation for TSDuck. If you propose to write one, you are welcome.

nightfurybd05, 2020-09-07:
Hello sir,
I am an  ⏩ Expert ⏪ logo design,PDF, scanned and any Document edit and Photoshop expert  A-Z, I can do any type of Photoshop editing for your Pics or Documents and other kinds of stuff such as driving license and certificate and any other Document you want full A-Z. unlimited revisions, and 100% money-back guarantee.
Whatsapp:+8801893523444
Fiverr:https://www.fiverr.com/night_fury_bd  
My work:https://www.behance.net/night_fury_bd

lars18th, 2022-01-25:
Hi @engineer507 ,

> Hello how i can add rtsp streaming in tsduck? is there any rtsp output plugin for tsduck? thanks robert

I've created a guide to do it with an external helper tool: https://github.com/tsduck/tsduck/issues/928#issuecomment-1021161514

So, I suggest to close this issue.
Regards.