tsduck/tsduck #474: Query: What's the quickest way with TSDuck to jump to a specific time in a transport stream?

mcflyuk, 2019-11-29:
When you have a long TS capture (say, half a day), I'd like to jump to a certain time in the stream before playback. Same kind of action as dragging the playout bar in Dektec's StreamXpress software.

I've found out it's possible to achieve with the -time --tdt plugin:

> -P time --tdt --drop _(time in first TDT section)_ --pass _(time to jump to)_

I was wondering whether this was the quickest method as it takes a few minutes before playback starts, seems like it's processing every packet before getting to the time to jump to.

Are there any alternative methods to achieve this kind of action that would be quicker?

lelegard, 2019-11-29:
If the input is a file, a direct access one, not a pipe, you may have a look at input plugin `file` options such as `--byte-offset` or `--packet-offset`. But this is a physical offset in the file, not a time reference. If you want a time reference from the file, then you need to read all packets, to get those references precisely. But if you know the file bitrate, it is relatively trivial to compute the packet offset of a given time offset from the beginning of the file.

If the input is not a file, then all other predefined input plugins handle sequential inputs. So you need to read all packets. If you wan to speed up the processing, you may use the plugin `skip` right after the input plugin. All packets are _read_ from the input but the first _N_ packets are immediately dropped. But I don't think you really gain anything.

mcflyuk, 2019-12-02:
Thanks for the suggestions. It will be file input so I will have a look at the options for the _file_ plugin.

The bitrate is fixed (4Mbps) and I know the duration of the transport stream so sounds like _--packet-offset_ may be the way to go.

lelegard, 2019-12-02:
You do not even need the TS duration, only the bitrate.

Using `bash` syntax:
~~~
BITRATE=4000000
START_SECOND=10

tsp -I file --packet-offset $(( ($BITRATE * $START_SECOND) / (188 * 8) )) ...
~~~

mcflyuk, 2019-12-02:
> You do not even need the TS duration, only the bitrate.
> 
> Using `bash` syntax:
> 
> ```
> BITRATE=4000000
> START_SECOND=10
> 
> tsp -I file --packet-offset $(( ($BITRATE * $START_SECOND) / (188 * 8) )) ...
> ```

That's great. I can use this to jump to an approximate position in the stream, then use the _-P time --tdt_ plugin to find the actual TDT time I'm looking to start playout from. Cuts down on parsing time a lot compared with using the time --tdt plugin from the beginning of the stream.

As a quick question before I close this ticket - do you have an idea of what date your next tsduck update will be released? So I'll know when to move over from my locally built tsp.exe to the proper release?

lelegard, 2019-12-02:
> do you have an idea of what date your next tsduck update will be released?

Version 3.19 was released one week ago. So, do not expect version 3.20 before a couple of months. You mention your "locally built tsp.exe". So, I assume that you use Windows and you are able to compile. The safest way is to build you own installer and use the latest TSDuck version from the repo. This is extremely simple. After [installing the prerequisites](https://tsduck.io/doxy/building.html#reqwindows) only once, building from scratch a complete set of binary installers is as simple as double-clicking on script `Build-Installer.ps1` (you have time to lunch then...)

mcflyuk, 2019-12-03:
OK, that's useful information, thanks for the instructions for building an installer.

closed 2019-12-03 by mcflyuk