tsduck/tsduck #475: Clarification regarding DVB-S2 multistream operation

igorauad, 2019-12-02:
Hi all,

Does anybody know if the DVB support in Linux is such that multiplexed DVB-S2 BBFRAME streams with different ISI can be received at the same time?

I understand that the `dvb` plugin of `tsp` supports specification of the input stream identifier ISI. However, could it specify multiple ISIs to be received at the same time? And would the kernel/OS support that? Also, is this supported by the DVB-S2 standard at all, that is, the concurrent reception of multiple streams with distinct ISI?

Thanks

lelegard, 2019-12-03:
This is clearly a question for the Linux DVB mailing list.

That being said, `tsp` processes a suite of TS packets which collectively create a consistent transport stream. If you receive TS packets from multiple independent streams, how would you rebuild a consistent stream? When receiving a packet with, say, PID 0, which PAT (or part of PAT) is this?

If the kernel allows receiving two independent streams which are demodulated from the same physical source, then the API shall be designed to discriminate between the two (maybe a different `dvr` device under the same `adapter`). In all cases, `tsp` will read only one of them.

This is different from T2-MI for instance where the two streams are not independent. One is _inside_ the other. `tsp` reads the outer TS and the plugin `t2mi` extracts the inner TS and _replaces_  the outer TS with the inner one. But, in no circumstances, one single `tsp` instance can handle two independent TS.

igorauad, 2019-12-10:
Thanks, @lelegard.

I'm experimenting with a DVB-S2 system with two types of receivers: Linux USB receivers and software-defined receivers. For the software-defined receiver, I'm experimenting with `tsp -P mpe`.  

I am particularly trying to understand more about the two layers of multiplexing, at MPEG TS level (by PID) and at BBFRAME level (by ISI). The goal is to transmit IP traffic over MPE, within MPEG TS. So the entire encapsulation chain is:

UDP -> IP -> MPE -> MPEG TS -> DVB-S2

On transmit side, I'm transmitting two UDP streams, which are distributed to two distinct MPEG TS and correspondingly DVB-S2 streams. The two schemes I'm exploring are summarised below:

**Scheme 1 (same ISI, different PID)**

Stream 1:   ISI 1; PID 1; 
Stream 2:   ISI 1; PID 2;

**Scheme 2 (different ISI, different PID)**

Stream 1:   ISI 1; PID 1; 
Stream 2:   ISI 2; PID 2;

Now, to the questions.

>  If you receive TS packets from multiple independent streams, how would you rebuild a consistent stream?

- With USB receivers, the approach I'm currently using is to instantiate two [dvb net interfaces](https://www.kernel.org/doc/html/v4.10/media/uapi/dvb/net-add-if.html#c.NET_ADD_IF). Each interface is configured to a different PID. 
- With the software-defined receiver (SDR) approach, the SDR application decodes the MPEG TS stream and feeds into `tsp -P mpe`. The stream has both PIDs mixed (multiplexed together), but the `mpe` plugin successfully extracts the UDP datagrams regardless. I launch it as follows:

```
-P mpe --pid 32-33 --udp-forward --local-address 127.0.0.1 -O drop
```

Everything works as desired on Scheme 1, i.e. if the ISI is the same. However, there is a specific reason why Scheme 2 (distinct ISI) would be actually advantageous in the system I'm exploring. 

Using Scheme 2, the SDR receiver feeding `tsp -P mpe` should remain OK. Nothing will change from the point of view of the `mpe` plugin, since it will continue to see the same MPEG TS stream (with multiplexed PIDs).

The real problem is on Linux USB receivers. What I wanted to accomplish in order to make Scheme 2 function is to be able to receive both streams (with distinct ISI) at the same time. This really isn't dependent on `tsp`. Instead, it depends on drivers and hardware as you suggested. However, my question really comes from the fact that it could be up to the user space application to configure this.

I've been trying to use `dvbv5-zap`. I tried to set the ISI equal to the [`NO_STREAM_ID_FILTER `](https://github.com/tbsdtv/linux_media/blob/42a1373e41d9e54922b13d88b9add637c31527a3/include/uapi/linux/dvb/frontend.h#L756), such that all ISIs were allowed to go through at the same time. With one of the devices I'm testing, this did allow two ISIs to go through, but one of the two streams was barely passing (the adapter was probably dropping data). The other stream was with normal rate.

So I tried to explore the `tsp -I dvb` plugin from tsduck to see if it could configure the demodulator/tuner differently. I asked when I was wondering whether multiple ISI could be configured at the same time. However, by looking at the kernel implementation, it does seem that `NO_STREAM_ID_FILTER` (i.e. ISI = 4294967295) is the only possibility of allowing multiple ISI.

I guess I really need to ask on Linux DVB group.

Many thanks for the help.

lelegard, 2019-12-12:
The central concept here is the "logical consistent transport stream". As soon as you have a consistent TS, TSDuck can work on it.

As far as I understand (which may not be the exact reality), multiple ISI - _independent_ stream identifiers - mean distinct logical TS being transported side by side on the same modulated carrier. Am I right? Any DVB-S2 specialist to confirm or correct me?

> However, by looking at the kernel implementation, it does seem that NO_STREAM_ID_FILTER (i.e. ISI = 4294967295) is the only possibility of allowing multiple ISI.

If I am right, you cannot get two ISI at the same time in the same pipe. Each TS packet with a given PID number belongs to a given TS (in a given ISI) and you need to know which one. Maybe you can demodulate two or more ISI at the same time and send each one to a distinct pipe (the `dvr` devices on Linux). But in that case, you have several TS sources and each source produces a distinct TS. It you want to use TSDuck on them, then one `tsp` instance would be needed for each ISI.

Assuming that some external program is able to demodulate 3 ISI on `/dev/dvb/adapter0`, then _maybe_ the 3 different independent TS would be readable from `/dev/dvb/adapter0/dvr0`, `/dev/dvb/adapter0/dvr1` and `/dev/dvb/adapter0/dvr2`. But this is purely hypothetical.

Would this be true, then you would need to run three instances of `tsp`, one on each `dvr` device.

>I guess I really need to ask on Linux DVB group.

Please let us know any valuable information you get from them.

As a comparison, MPE and T2-MI are quite different from multiple ISI. With MPE or T2-MI, the two TS are not _independent_. One TS - the inner one - is encapsulated inside one PID of the other TS - the outer one. You may have multiple inner TS in different PID's of the outer TS. You may even have several inner TS in the same PID of the outer TS (with different IP addresses on MPE or different PLP on T2-MI).

So, you first demodulate the outer TS and then you decapsulate another TS  from a PID. Using `tsp` plugins such as `mpe` or `t2mi`, you can extract one - and only one - inner TS. You may select to process the inner TS as a _replacement_ of the outer TS or just save it away. But `tsp` cannot process the two TS at the same time.