tsduck/tsduck #479: MSVC Compiler error C5031 when transitioning from 3.17 to 3.19

khukman, 2019-12-05:
I am writing new code that requires the use of the new TSFile class (allows read and write from the same open).  When I compiled in the new version, I am getting the following MSVC error:

2>c:\program files\boost\boost_1_58_0\boost\move\detail\config_end.hpp(11): error C5031: #pragma warning(pop): likely mismatch, popping warning state pushed in different file (compiling source file S:\devel\adr\RADAS\trunk\software\programs\asset_interface_c++\common\MyFileSink.cpp)

My code has not changed at all, and when I revert to the 3.17 version, the issue is not present.  Any help would be greatly appreciated.

lelegard, 2019-12-05:
In recent versions of TSDuck, all compilation warnings are enabled. This is good to track tiny little nasty bugs.

This works fine with gcc and clang. But, as usual, Windows is creating a mess when you try to work cleanly. The official headers from Microsoft generate tons of warnings from their own compiler. So, we have to push/pop a lot of warnings before including their headers. And then, comes the _second_ layer of bugs in Microsoft headers: in one of the system headers, they do their own push/pop of warnings but they do it incorrectly and forgot to pop their own push. So, when you do _your_ pop, you pop in fact the push of the system header. See [mode details here](https://developercommunity.visualstudio.com/content/problem/387055/warning-c5031-pragma-warningpop-likely-mismatch-po.html). This explains the error.

Now, about your problem, I see that you compile your application, not TSDuck itself. I suspect that the boost headers have the same problem. The header file which generates the error is named `config_end.hpp`. Maybe the warning push is done in some `config_begin.hpp` and the pop in `config_end.hpp`. While legal, this is a bad practice to push and pop warnings in different files.

You may try to add this before including the boost headers:
~~~
#if defined(_MSC_VER)
#pragma warning(disable:5031)
#pragma warning(disable:5032)
#endif
~~~

ZacharyJoswick, 2019-12-05:
I am khukman's coworker and have been working on this issue with him. We finally resolved the issue after adding the pragma warning definitions as you stated after including the tsduck headers. 

The issue appears to stem from from the tsplatform.h file. In this file at line 906 warning 5031 is unsuppressed and declared as an error. This appears to be the source of our issue. It appears that in commit c9fd251d4d9adcc8034233a8bb933ce861b997e2 these lines are removed. Upon removing these lines ourselves we can confirm that the issue does not appear. We will use the workaround for now. Thank you for your help.

closed 2019-12-06 by khukman