tsduck/tsduck #480: Query: Is there an equivalent method to tsp -I file --packet-offset but for the end of the file?

mcflyuk, 2019-12-05:
This is related to https://github.com/tsduck/tsduck/issues/474 but still a separate query.

I'm now using _-I file --packet-offset XXXXXX_ to jump to the nearest packet in a file before starting looking for a TDT packet matching the required time (_-P time --tdt --drop HH:MM:SS --pass HH:MM:SS_). This cuts down on processing time a lot.

There's now an additional scenario where having processed the required period in the stream, the remaining duration of the stream isn't required. So I instruct tsp to do an additional drop from the end time (_-P time --tdt --drop HH:MM:SS --pass HH:MM:SS --drop HH:MM:SS_). But the similar scenario of processing all the TDT packets comes up again, only this time the remainder packets through to the end of the file. 

Effectively what I'm trying to describe is if I want to take a period from the middle of a transport stream, it'll be quickest to jump to a location by packet number, but then also to drop all the packets after another packet number. Something like _-I file --packet-offset XXXXXX --packet-offcut YYYYYY_.

Is this functionality already existing in TSDuck and I've not yet found how to do it in the documentation? Or is it something new that could be forked off the _--packet-offset_ code?

lelegard, 2019-12-05:
Hi @mcflyuk 
The plugin `until` is here to terminate the processing at a specific condition. Especially, `-P until --packets NNN` terminates after NNN packets. So, if you know the size of the file and the initial packet offset, you know after how many packets you should terminate.

mcflyuk, 2019-12-05:
Great, such a quick response!

Thanks for the information, I'll give that a try and report back.

mcflyuk, 2019-12-12:
Looks like the _-p until_ plugin is working, hard to confirm if the time is much different but the overall performance is still a big increase so it's working well, closing off.

closed 2019-12-12 by mcflyuk