tsduck/tsduck #481: Issue: -P timeref updates TDT and TOT dates, but doesn't update the TOT local_time_offset/time_of_change/next_time_offset

mcflyuk, 2019-12-12:
I've found through my testing that using _-P timeref_ to change the date of a transport stream, the date is updated in the TOT correctly.

However the local_time_offset/time_of_change/next_time_offset fields in the TOT still stay the same.

Because of the two TOT changes in a year (forwards/back), if you are changing a stream from a winter date to a summer date the local_time_offset would change between 0 and 1.

From a UK perspective the time_of_change could also be re-calculated and updated. The last Sunday in March the clocks always go forward one hour, the last Sunday in October the clocks always go back one hour.

I guess because of all the different rules for country's times around the world, this would make it harder to implement. Has it been considered before?

lelegard, 2019-12-12:
Only the UTC time is changed. Changing the local time offset (and next time of change) is not implemented. But it could be, this is a good idea. This is not particularly difficult to implement when you know what to set.

The difficulty is to know what to set. As you mention, the rules depend on the country and the rules change over time. In Europe, there is a current discussion about the usefulness of daylight saving time (DST). Completely removing DST is an option, thus eliminating the need to adjust the local time offset depending on the period of the year. But I am referring to Europe, you mentioned UK (sorry, I couldn't resist 😃).

So, it seems more reasonable to explicitly specify new values on the command line for any of :
- local time offset
- time of change
- next local time offset after change
- country code and/or region id filtering (default: all)

mcflyuk, 2019-12-12:
>  But I am referring to Europe, you mentioned UK (sorry, I couldn't resist 😃).
Hey, we're still part of Europe! Can't say whether that will be the case by the time of the next TSDuck release, who knows?

> So, it seems more reasonable to explicitly specify new values on the command line for any of :
> 
> * local time offset
> * time of change
> * next local time offset after change
> * country code and/or region id filtering (default: all)

Yes, for the UK at least it's very straightforward for calculating next offset and what value it will be. But if you're supporting transport streams from around the world, allowing a manual input of the values would be a preferable approach.

It would be even better to abandon daylight savings and stick to British Summer Time. But every year it gets suggested and every year the Scots complain they don't get any sunlight in the mornings so it keeps going.

lelegard, 2019-12-12:
> But if you're supporting transport streams from around the world, allowing a manual input of the values would be a preferable approach.

OK, will do...

> It would be even better to abandon daylight savings and stick to British Summer Time. But every year it gets suggested and every year the Scots complain they don't get any sunlight in the mornings so it keeps going.

This problem will go away some time after the Brexit (thus possibly meaning never). After the British exit a larger community they were once not part of, there is no reason to prevent the Scots for doing the same thing. Sorry, since transportation strikes are paralyzing my side of the Channel, it is a relief to make fun of what is going on on the other side of it 😄 😄

lelegard, 2019-12-12:
Done. Added options `--local-time-offset`, `--next-change`, `--next-time-offset`, `--only-country` and `--only-region` to plugin `timeref`.

closed 2020-01-13 by lelegard

mcflyuk, 2020-03-25:
I've been doing some testing with the new build now, thanks for the update.

On the whole it works quite well. I can now set all the values in the local_time_offset_descriptor and I've implemented code to calculate when the next UK time zone change takes place.

However if you put a time_of_change (--next-time-offset) into a transport stream and it's scheduled to happen within the duration of the stream, the local_time_offset doesn't get updated to the new offset (that was set in next_time_offset) at this time.

Is this functionality something that could be included?

mcflyuk, 2020-03-25:
[image]

I've taken a screengrab to try to show what I mean. Excuse the handwriting from the snipping tool!

lelegard, 2020-03-25:
> I've taken a screengrab to try to show what I mean. Excuse the handwriting from the snipping tool!

To show the content of tables and sections here, it is more convenient to use `tstables` or the plugin `tables`. The output is text, easier to comment and post than a GUI screenshot.

> However if you put a time_of_change (--next-time-offset) into a transport stream and it's scheduled to happen within the duration of the stream, the local_time_offset doesn't get updated to the new offset (that was set in next_time_offset) at this time.

This is more complicated than that. Once you cross the "next time of change", everything needs to be updated, including the "next time of change". So, you need an option to specify the "next time of change after next time of change", etc. So, even if we update the local time offset, the section in your example remains incorrect because the next time of change if lower than or equal to the current date.

A real-world production tool would need a complete schedule of time of change / time offsets or an algorithm to generate them. Correctly updating the TOT on the long run deserves a full specification.

TSDuck is just a lab tool to run tests and demos.

mcflyuk, 2020-03-25:
OK, thanks for the reply, not to worry. We can get round it by generating two streams, one before and one after the time change. But I thought I would mention it anyway.