tsduck/tsduck #482: Request: 1-In-1-Out feature

lars18th, 2019-12-13:
Hi @lelegard ,

The current functionalities of the TSDuck are impresive. I would like to thank you for the scale of this project. I comment this because new functionalities can open new uses of it.

I'm not sure if this can be done now or not. But I want to use the `tsp` for the purpose that I will describe now. I prefer to explain the precise use case as I feel you will understand it faster:

- Inside another process I want to _process_ just one pid (for example, the 0x16 EIT pid, or the 0x20 TOT/TDT, or whatever) with the `tsp` tool. So the idea is to capture _every_ single packet inside the process and pass a copy of it to the `tsp` process over a pipe. Then receive a new TS packet over the pipe and replace the original packet with the received one.

However, what's the problem here? The problem is that the `tsp` process is oriented to realtime or off-line chunk processing. So a _group_ of packets are processed at some time. But the idea here it's a running mode with 1-In-1-Out TS processing mode. This mode will grant that for every single input packet a new single packet will output. For sure, some time for the processing will be consumed; but this will a problem for the master process, and not for the `tsp` slave process. Futhermore, I'm not requesting to "advance in time", as it's impossible. So when the processing will starts then the output packets will be NULL packets until the internal processor can output real packets. This doesn't needs to be a real problem.

What do you think of this idea? Is it feasible? Or can it be done now with the current implementation?

lelegard, 2019-12-13:
Everything is possible. This is just a matter of development. In this case, the packet processing is done in a separate process. I assume that you develop this custom application. Why not developing it directly as a plugin? If you need to communicate with another application of yours, then this is your private protocol.

lars18th, 2019-12-13:
Hi @lelegard ,

Thank you for the response.

Regarding the App, is not my development. I want to add to some opensource tools (for example, SAT>IP servers and DLNA servers) the option to rewrite/add some PSI tables. In this case it's quite simple to open a child process for an _external processor_ and send to it each single TS packet and receive a new one. This will work even if the main process is writen in a different language (not C/C++). And especially when it comes to scripts that use different tools.

I hope you will understand the target of the idea and the benefit of such operation mode.
In any case, thank you for your support!
Regards.

lars18th, 2019-12-13:
Hi @lelegard ,

Two additional comments to be more clear:

1. _"and send to it each single TS packet"_ in this phrase "each" refers to each packet of a given pid. Not each TS packet of the stream. The objective is to only process low bit rate pids, such EIT for example.
2. _"the packet processing is done in a separate process"_ regarding this, take note that the processing is done inside the master process. The "external processor" will be used only for the small part of specific pids.

In any case, let me ask you directly: Do you have a better idea about this objective?

lars18th, 2019-12-19:
Hi @lelegard ,

Any idea to _aproximate_ to this behaviour with _current_ implementation?
Perhaps adjusting some buffer sizes.

I hope you want to comment about it.
Thank you!

lelegard, 2020-03-27:
Duplicate of #553, now fixed

closed 2020-03-27 by lelegard