tsduck/tsduck #483: Question: Detecting I-frames starting GOPs

lars18th, 2019-12-13:
Hi @lelegard ,

Another crazy request: Sometimes someone has requested about how to create loop videos with TSDuck. As you know the main problem is to detect the start of the video to not cut before or after the first I frame of a GOP. And I know that this is the responsability of a video processor, and not for a library like TSDuck that only processes MPEG-2 TS data.

However, it will be useful to have some plugin that can detect the start of a GOP of some currently used video codecs (MPEG-2, H.264, HEVC). The detection of the PUSI mark it's already included. But with some additional processing perhaps it will not very hard to detect the start of the GOP checking the first PES packet of a video pid when a PUSI mark is set.

For sure this will interesting too for SCTE-35 processing. And I feel the HLS output plugin has some segmenter that has the required functionality.

You think this has sense, or it's very complex?

lelegard, 2019-12-14:
In the old days of MPEG-2 video, a PES packet in a video PID contained exactly one GOP, typically one every 500 ms. So, a TS packet with PUSI always meant the start of a GOP.

Nowadays, with codecs such as AVC or HEVC, I am not sure. We would need the advice of a video codec specialist. Maybe you could ask the question in some some video codec forum.

I assume that the start of a GOP will always be at the start of a PES packet because an I-frame in the middle of a PES packet would clearly be a loss of opportunity to resynchronise on a channel switch for instance. I do not say that having a GOP starting in the middle of a PES packet is impossible, I just say it would be stupid.

On the opposite, the real question is: is it possible to have a PES packet starting in the middle of a GOP and how can we detect that? I do not have the answer and I assume it would be codec-specific.

lars18th, 2019-12-16:
Hi @lelegard ,

For sure the solution is codec-specific. Why? Because every video PES packet contains only one frame. And to obtain the START of the GOP we need to analyze the PES content; as each PUSI mark only indicates the start of a _FRAME_, and not of a _GOP_.

So the plugin will do:
- For each PUSI mark check if the content of the PES packet is a GOP start.

And to complete this based on used codec:
- For MPEG-2: Parse the first PES packet data with the Sections Data. If a "Group Of Pictures" is found (type 0xB8) then is the start of the GOP; and if is not found then is not a GOP start.
- For H.264/AVC: In this case we need to search for an IDR I-frame or non-IDR frame. So parse the PES packet for a NAL type 5 (Coded slice of an IDR picture) or NAL type 1 (Coded slice of a non-IDR picture).

Then for a loop video, will be responsability of the user to use closed-GOPs to overcome decoding errors at the loop point. And if this it's true, then the filter will catch the exact point where cut-off.

You agree?

lelegard, 2019-12-16:
> Because every video PES packet contains only one frame.

Not always. For live MPEG-2 video, we used to have one complete GOP (ie. 12 frames) in a PES packet. But I agree that in all recent streams I have found as many PES packets per second as frames.

Otherwise, what you are suggesting is doable. But there is no guarantee that the start of I-frame can be found in the first TS packet of the PES packet. So, you need to start demuxing the PES packet and when you realize that the PES packet contains an I-frame, then the first TS packet for this PES packet may be in the past. This can be a problem or not, depending on what you want to do.

lars18th, 2019-12-16:
> But there is no guarantee that the start of I-frame can be found in the first TS packet of the PES packet. So, you need to start demuxing the PES packet and when you realize that the PES packet contains an I-frame, then the first TS packet for this PES packet may be in the past.

Now I see the problem. You're rigth! The TSDuck library process _packets_ only. So when checking a PUSI packet, it requires to confirm if it's a starting point of a GOP. And it's very uncommon that in the first packet it's all the required info.

Any idea?

lelegard, 2019-12-16:
> Any idea?

Sorry, I cannot see in the future ;) So, from a PUSI packet, I cannot guess if some forthcoming TS packet on the same PID will contain a start of I-frame (in the near future). 

Even with a different software architecture, you need to buffer the complete TS to make sure that a PUSI packet is not output as long as you cannot confirm that the corresponding PES packet contains (or not) an start of I-frame. So, you introduces some latency.

As far as I understand, SCTE-35 systems "guess" the PTS of a future event based on prior knowledge of the frame rate, GOP size and PTS of previous I-frames. You could do the same thing but you must acquire prior knowledge of the frame rate, GOP size and PTS of previous I-frames. Then since the PTS is usually (not guaranteed to be though) in the same TS packet as the PUSI, then you can guess if a PUSI packet is likely the start of an I-frame. But this become a Nostradamus-like system...

lars18th, 2019-12-16:
> But this become a Nostradamus-like system...

:laughing: Not really! In fact it's only a two-pass process: In the first pass you check the PES packets, and insert some mark when you found the GOP start. And in the second pass you believe the mark. Futhermore, for a realtime processing the two-pass can be done with a buffering in between with the `timeshift` plugin.

Any idea to the best mode to mark the exact point of the GOP start? For example, some tricky mark in the TS header where the PUSI mark is found? Or a custom SCTE-35 mark inserted in the stream with the PCR of the packet with the PUSI mark?

lelegard, 2019-12-17:
> In fact it's only a two-pass process: 

Precisely. So, it does not match the `tsp` model which is a streaming processor. This kind of processing needs a specific application. Feel free to develop it.

lars18th, 2019-12-17:
> Precisely. So, it does not match the `tsp` model which is a streaming processor. This kind of processing needs a specific application.

:stuck_out_tongue_winking_eye: `tsp ... -P timeshift | tsp -P gop-detector ...`

Please, don't worry about it, it's a simple joke!

The question is: _Any idea about the best mode to mark the exact point of the GOP start?_

The problem is how to _share_ the information between the first process and the second process. In fact, inside the current implementation of TSDuck some plugins parse multiple packets (for example, plugins that read tables). So the _detection_ plugin can be implemented. However, I need help about the best solution to share this information with the second process.

You have any idea?

lars18th, 2019-12-24:
Hi,

Since the detection of the I-frames inside the PES content it's outside of the TSDuck work, I close this issue.

A different approach is suggested in #492 .

closed 2019-12-24 by lars18th