tsduck/tsduck #484: tsswitch fork option

enricovittorini, 2019-12-17:
Hello gents,
i was approaching tsswitch but had a problem using "fork"
```
tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.19-1520

tsswitch -I fork 'tsp -I file test.ts'
* Error: fork[0]: error creating process: The system cannot find the file specified.
* Error: file[1]: cannot open test.ts': The system cannot find the file specified.

```
of course test.ts file is there.

lelegard, 2019-12-17:
You did not mention which system. From the system error message, I guess that this is Windows.

Which command interpreter are you using? DOS, PowerShell, bash, other? I guess DOS. Because the problem is here: you have a command interpreter problem, not a TSDuck problem.

Please carefully inspect the error messages. The answer is there.
~~~
tsswitch -I fork 'tsp -I file test.ts'
* Error: fork[0]: error creating process: The system cannot find the file specified.
* Error: file[1]: cannot open test.ts': The system cannot find the file specified.
~~~

First, `error creating process` means that the process is not created, the command is not found. So `tsp` does not run and does not even try to open the TS file.

Second, the message about `test.ts` is prefixed by `file[1]`. Having `[1]` means that the error is reported by the second (index 1) input plugin of `tsswitch`, `tsp` would not add an index.

Third, look closely at the `cannot open` message. It says `test.ts'`, not `test.ts`. Do you see the difference? The trailing quote?

So, my conclusion is the following. You are using a command interpreter which, for some reason, does not recognize the quotes as most civilized command interpreters.

The scenario is:
- The `fork` plugin tries to create a process with parameter-less command `'tsp` (with a leading quote). There is no such program of course => `error creating process`.
- The `-I file` is interpreted as `tsswitch` _second_ input plugin, not a `tsp` plugin.
- And it tries to open file `test.ts'` (with a trailing quote). There is no such file

Indeed, if I open a DOS command window (something I _never_ do):
~~~
C:\> tsswitch -I fork 'tsp -I file test.ts'
* Error: fork[0]: error creating process: The system cannot find the file specified.
* Error: file[1]: cannot open test.ts': The system cannot find the file specified.

C:\> tsswitch -I fork "tsp -I file test.ts"
* Error: file: cannot open test.ts: The system cannot find the file specified.
~~~
In the second case, the error comes from the created `tsp` process, not `tsswitch` (there is no index in the plugin name) and the error mentions `test.ts` without trailing quote (the file does not exist on my system).

Apparently, the DOS command interpreter accepts only double quotes, not simple quotes.

Lessons learned:
- _Carefully_ read error messages, pay attention to details.
- Do not use an idiotic rudimentary obsolete 40-years old command interpreter such as DOS.

lars18th, 2019-12-17:
Hi @lelegard ,

> * Do not use an idiotic rudimentary obsolete 40-years old command interpreter such as DOS.

I'm sorry! I appreciate you as a very good person. However, I need to comment my point of view about this:
- I use Windows and Linux, and not MacOS.
- Inside Windows I use CMD, but Cygwin Bash and WSL Bash too (depending if it's Win7 or Win10).
- I never use PowerShell (I don't like it!).
- The CMD shell is not DOS. And from my point of view this simple shell it's preferable over PowerShell. So, if I need more than the CMD shell then I'll use Bash (from Cygwin or WSL).

Then, I feel it's preferable to leave to the user/developer to use whatever shell... if it's possible to support it. And at time the CMD shell in Windows is supported, rigth?
I undertand that this issue is related to the shell and not with the library. However, please don't remove support for the _good_ CMD shell. Some of us want to keep using it.

Regards.

lelegard, 2019-12-17:
> However, please don't remove support for the good CMD shell.

I won't remove that support simply because there is nothing to remove. TSDuck does not provide "support" for any kind of shell. TSDuck, just like any other program, just needs to receive the correct command line parameters.

By typing `tsswitch -I fork 'tsp -I file test.ts'`, your intent was to pass those parameters:
~~~
-I
fork
tsp -I file test.ts
~~~

But the parsing of CMD (not better than DOS, sorry) is such that `tsswitch` receives those:
~~~
-I
fork
'tsp
-I
fil
test.ts'
~~~

lars18th, 2019-12-17:
> > However, please don't remove support for the good CMD shell.
> 
> I won't remove that support simply because there is nothing to remove.

Great! :wink: 

> By typing tsswitch `-I fork 'tsp -I file test.ts'`, your intent was to pass those parameters:
> But the parsing of CMD (not better than DOS, sorry) is such that `tsswitch` receives those:

I know it! All CMD users use double quotes, since single quotes have never been supported in this shell. For this reason in _all_ Windows examples the quotes are `"`.

Regards.

enricovittorini, 2019-12-17:
@lelegard switch to cygwin. this is ok
thanks

closed 2020-03-04 by lelegard