tsduck/tsduck #486: How to extract all data from TS file

phamduythai92mta, 2019-12-18:
Hi all!
i am absolutely new to TSDuck. could tyou kindly hepl me with the quention. I have a file.ts, i wanna get all the data from this file. how can i do that?  TSduck is able to do this?
thanks in advance!

lelegard, 2019-12-18:
It depends on what kind of "data" you are talking about. You may start with the `tsanalyze` command which produces an overview of the transport stream structure. If you need more "data", then you should RTFM. There is such a big user's guide because there are many possibilities.

phamduythai92mta, 2019-12-23:
thank you for quick reply!
sorry about stupid questions, however i am totally new to this field, and there are an enormous number of resources and i don't know where i should start.

https://drive.google.com/open?id=1Fy7f9p5OoqrfWqqX9D6kk4ph0qP3asab

that is the file after i use ts analyze my ts file which i got from the wild. i wanna get video or audio or text data from this file. so could you kindly tell me what i should do and which plugins should be applied?
many thanks!