tsduck/tsduck #487: Not building with gcc

joolzg, 2019-12-19:
gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609

I know its old but the machine this is on cannot be upgraded so any workaround?

In file included from base/tsIntegerUtils.h:124:0,
                 from base/tsForkPipe.cpp:37:
base/tsIntegerUtilsTemplate.h: In function 'INT ts::SignExtend(INT, int)':
base/tsIntegerUtilsTemplate.h:153:24: error: unknown option after '#pragma GCC diagnostic' kind [-Werror=pragmas]
         TS_GCC_NOWARNING(shift-negative-value)
                        ^
  [CXX] base/tsInputRedirector.cpp
In file included from /home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtils.h:124:0,
                 from plugin/tsDektecInputPlugin.cpp:35:
/home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtilsTemplate.h: In function 'INT ts::SignExtend(INT, int)':
/home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtilsTemplate.h:153:24: error: unknown option after '#pragma GCC diagnostic' kind [-Werror=pragmas]
         TS_GCC_NOWARNING(shift-negative-value)
                        ^
  [CXX] base/tsIntegerUtils.cpp
  [CXX] base/tsInterruptHandler.cpp
In file included from /home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtils.h:124:0,
                 from plugin/tsDektecOutputPlugin.cpp:37:
/home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtilsTemplate.h: In function 'INT ts::SignExtend(INT, int)':
/home/encoder/src/tsduck/src/libtsduck/base/tsIntegerUtilsTemplate.h:153:24: error: unknown option after '#pragma GCC diagnostic' kind [-Werror=pragmas]
         TS_GCC_NOWARNING(shift-negative-value)
                        ^
  [CXX] dtv/tsKeyTable.cpp
  [CXX] dtv/tsLNB.cpp
  [CXX] dtv/descriptors/tsLinkageDescriptor.cpp
  [CXX] dtv/descriptors/tsLocalTimeOffsetDescriptor.cpp
cc1plus: all warnings being treated as errors
../../Makefile.common:903: recipe for target 'release-x86_64/tsForkPipe.o' failed
make[2]: *** [release-x86_64/tsForkPipe.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from base/tsIntegerUtils.h:124:0,
                 from base/tsIntegerUtils.cpp:34:
base/tsIntegerUtilsTemplate.h: In function 'INT ts::SignExtend(INT, int)':
base/tsIntegerUtilsTemplate.h:153:24: error: unknown option after '#pragma GCC diagnostic' kind [-Werror=pragmas]
         TS_GCC_NOWARNING(shift-negative-value)

joolzg, 2019-12-19:
Ive commented out the pragma and all builds so maybe just needs a version guard around it

lelegard, 2019-12-20:
Fixed in commit c32326888d8e68534866330300ea7ce009c178e4

closed 2019-12-25 by joolzg