tsduck/tsduck #488: HLS input: probably incorrect URL compilation

alexander-nf, 2019-12-20:
Hello Thierry,

I am unable to get a HLS-stream and the reason, I think, is obvious, m3u8-playlists contain relative paths to playlists and chunks. As a result, the paths are concatenated and doubled in URLs TSDuck (3.19-1520) requests. Please look at the debug information below.

The master playlist request:
```
$ tsp -d1 -I hls --live http://public.internet.server/channel/Animals.m3u8 -O drop

tsp: Debug: debug level set to 1
* Debug: debug level set to 1
tsp: Debug: trying to load "/usr/bin/tsplugin_hls.so"
tsp: Debug: symbol tspNewProcessor not found in /usr/bin/tsplugin_hls.so
tsp: Debug: trying to load "/usr/bin/tsplugin_drop.so"
tsp: Debug: symbol tspNewInput not found in /usr/bin/tsplugin_drop.so
tsp: Debug: symbol tspNewProcessor not found in /usr/bin/tsplugin_drop.so
tsp: Debug: realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: hls: debug level set to 1
* Debug: drop: debug level set to 1
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading http://public.internet.server/channel/Animals.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.2
* Debug: hls: HTTP header: Date: Thu, 19 Dec 2019 14:38:41 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegURL
* Debug: hls: HTTP header: Content-Length: 112
* Debug: hls: announced content size: 112 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Cache-Control: no-cache, no_cache
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Last-Modified: Thu, 19 Dec 2019 14:38:38 GMT
* Debug: hls: HTTP header: Expires: Thu, 19 Dec 2019 14:38:40 GMT
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* hls: downloaded Animals.m3u8, master playlist, static, 1 media playlists
* hls: selected playlist: /channel/Animals/4k.m3u8, 3840x2160, 25,571,632 b/s
* Debug: hls: downloading http://public.internet.server/channel//channel/Animals/4k.m3u8
* Debug: hls: HTTP header: HTTP/1.1 404 Not found
* Debug: hls: HTTP header: Server: nginx/1.12.2
* Debug: hls: HTTP header: Date: Thu, 19 Dec 2019 14:38:41 GMT
* Debug: hls: HTTP header: Content-Length: 0
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: MIME type:
* Error: hls: invalid HLS playlist, does not start with #EXTM3U
* Error: hls: no more media playlist to try, giving up
* Debug: Report logging thread terminated
```

The content of the master playlist http://public.internet.server/channel/Animals.m3u8:
```
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:BANDWIDTH=22951664,RESOLUTION=3840x2160
/channel/Animals/4k.m3u8
```

I also did the media playlist request:
```
$ tsp -d1 -I hls --live http://public.internet.server/channel/Animals/4k.m3u8 -O drop

tsp: Debug: debug level set to 1
* Debug: debug level set to 1
tsp: Debug: trying to load "/usr/bin/tsplugin_hls.so"
tsp: Debug: symbol tspNewProcessor not found in /usr/bin/tsplugin_hls.so
tsp: Debug: trying to load "/usr/bin/tsplugin_drop.so"
tsp: Debug: symbol tspNewInput not found in /usr/bin/tsplugin_drop.so
tsp: Debug: symbol tspNewProcessor not found in /usr/bin/tsplugin_drop.so
tsp: Debug: realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: hls: debug level set to 1
* Debug: drop: debug level set to 1
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading http://public.internet.server/channel/Animals/4k.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.2
* Debug: hls: HTTP header: Date: Thu, 19 Dec 2019 14:42:22 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegURL
* Debug: hls: HTTP header: Content-Length: 942
* Debug: hls: announced content size: 942 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Cache-Control: no-cache, no_cache
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Last-Modified: Thu, 19 Dec 2019 14:42:19 GMT
* Debug: hls: HTTP header: Expires: Thu, 19 Dec 2019 14:42:21 GMT
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* hls: downloaded 4k.m3u8, media playlist, updatable (live), 16 segments, 5 seconds/segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: dropping initial segment
* Debug: hls: internal push-input thread started
* Debug: hls: downloading segment http://public.internet.server/channel/Animals//channel/Animals/4k/1576766530794.ts
* hls: downloading from http://public.internet.server/channel/Animals//channel/Animals/4k/1576766530794.ts
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 404 Not found
* Debug: hls: HTTP header: Server: nginx/1.12.2
* Debug: hls: HTTP header: Date: Thu, 19 Dec 2019 14:42:22 GMT
* Debug: hls: HTTP header: Content-Length: 0
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: downloading http://public.internet.server/channel/Animals/4k.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.2
* Debug: hls: HTTP header: Date: Thu, 19 Dec 2019 14:42:22 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegURL
* Debug: hls: HTTP header: Content-Length: 942
* Debug: hls: announced content size: 942 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Cache-Control: no-cache, no_cache
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Last-Modified: Thu, 19 Dec 2019 14:42:19 GMT
* Debug: hls: HTTP header: Expires: Thu, 19 Dec 2019 14:42:21 GMT
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: playlist media sequence: old: 76/0, new: 60/16
* Debug: hls: no new segment in playlist
^C
```

The content of the media playlist http://public.internet.server/channel/Animals/4k.m3u8:
```
#EXTM3U
#EXT-X-MEDIA-SEQUENCE:314
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:9
#EXTINF:4.000,
/channel/Animals/4k/1576763144497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763148497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763152497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763156497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763160497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763164497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763168497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763172497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763176497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763180497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763184497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763188497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763192497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763196497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763200497.ts
#EXTINF:4.000,
/channel/Animals/4k/1576763204497.ts
```

I don't know if these playlists are prepared correctly (HLS-compliant?) or not, but VLC and a few TV-set browsers are able to handle such playlists. Is it the TSDuck issue or should I use some workaround, scripting, etc.?

P.S. some additional information:
I don’t manage the internet web server, the stream. The received TS is signed by the “Streambuilder.pro Standard” software.

lelegard, 2019-12-20:
>  m3u8-playlists contain relative paths to playlists and chunks.

Most playlists contain relative paths and it works. Here, the paths are not relative to the playlist, they are relative to the host (meaning absolute path on the server). This is unusual but valid. So, this is a TSDuck problem. I will check and fix it.

I am not able to test your playlist because http://public.internet.server/ is not a public domain, it must be some internal domain of your organization (or "public.internet.server" is some local redirection in your organization). Do you have a public domain for this channel?

lelegard, 2019-12-20:
I setup a server with your playlists and fake TS segments. I confirm the problem. I just fixed it. You can rebuild TSDuck from the latest repo to get the fix.

alexander-nf, 2019-12-22:
I'm sorry, I misguided you and spent your extra time. The server is published on the internet, but I decided to change the server name in the logs, because I'm not sure I have right to share the content widely (from the point that the link will be indexed by web crawlers) and haven't got a response from the content owner yet.

I've just rebuilt TSDuck and can confirm now the new build has fixed the problem.
Thank you!

closed 2019-12-22 by alexander-nf