tsduck/tsduck #489: Playback file at original transmission and rendering speed

moffitj, 2019-12-21:
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.20-1565

I'm trying to use the sample with embedded SCTE-35 data (https://tsduck.io/streams/?name=russia-t2mi-scte35), but playback through tsp is at disk speed.

The user guide suggests that recomputing the bitrate with pcrbitrate and then using the regulate plugin should resolve this, but the output still comes out at disk rate:

tsp -r -I file /test/scte35samp.ts -P zap 800 -P pcrbitrate -P regulate -O ip 234.0.0.2:2000

I also tried setting a bitrate on the regulate plugin, but ultimately I don't know what the source bitrate originally was.

Am I missing something that would allow me to view the output in a video player like VLC or ffplay?

thank you.

lelegard, 2019-12-21:
Which TS does file scte35samp.ts contain?

Does it contain the full stream from the repository? I suspect that this is the case. This TS does not contain any viewable service. It only contains one big PID (0x1000) which encapsulates three distinct TS using T2-MI encapsulation. Since the outer TS does not contain any playable audio or video, it does not contain any PCR either. Consequently, the plugin `pcrbitrate` cannot compute any bitrate and the plugin `regulate`cannot regulate anything until it gets some valid bitrate, which will never happen.

If you want to view something with VLC, you need to extract one of the three encapsulated TS and then you need to select one service inside the decapsulated TS. This can be done in sequence with a command like this:
~~~
tsp -I file scte35samp.ts \
    -P t2mi --pid 0x1000 --plp 0 \
    -P zap 0x03F2 \
    -P pcrbitrate \
    -P regulate \
    -O ip 234.0.0.2:2000
~~~

I let you select another PLP (encapsulated TS) and another service id or name if you like.

To start with a local test, you may try the following command. It pipes the output service into a local media player (typically VLC) so that you can see what you are extracting. No need to regulate here, the input pipe of the media player will do it for you.
~~~
tsp -I file scte35samp.ts \
    -P t2mi --pid 0x1000 --plp 0 \
    -P zap 0x03F2 \
    -O play
~~~

moffitj, 2019-12-26:
Thank you. That worked. I misunderstood what the file was and didn't bother checking.

closed 2019-12-31 by lelegard