tsduck/tsduck #491: HLS with separate audio and video

enricovittorini, 2019-12-23:
Is there a way to get both the audio and video from HLS input when the are in two distinct playlist ( hlsv4),?

lelegard, 2019-12-23:
Not with TSDuck. Remember that TSDuck is a transport stream processor and the hls input plugin is just another way to read a transport stream.

HLS can be used to receive several types of data or combinations of data. The hls input plugin can only receive a TS. 

When audio and video are dissociated, the audio is usually not a TS. What you need is a generic HLS tool and a true multiplexer. TSDuck is none of these.

synologic, 2021-10-14:
Wow, two years in and nobody commented :)
Audio variants in HLS with TS container, are TS indeed.
This will not help if CMAF is used, then it's in MP4 container.

/opt/tools/ffmpeg/ffprobe -i https://bitdash-a.akamaihd.net/content/MI201109210084_1/audio/1_stereo_128000/hls/segment_12.ts
....
Input #0, mpegts, from 'https://bitdash-a.akamaihd.net/content/MI201109210084_1/audio/1_stereo_128000/hls/segment_12.ts':
  Duration: 00:00:03.97, start: 47.616000, bitrate: 172 kb/s
  Program 1
  Stream #0:0[0x101]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 127 kb/s