tsduck/tsduck #495: Continuous UDP stream analysis does not report correctly on ts bitrate changes

yaneti, 2020-01-02:
Hi,

tsduck 3.19 on Fedora.

Below is ~2min timestamped output  of `tsp  -I ip -r --default-interface  224.10.10.10:1234 -P analyze -i8 --normalized -O drop  | grep ^ts`

You'll notice that sometime around the halfway point the stream changes bitrate from 5Mbps to 4.5Mbps (this is done in a CBR padding MUX).
While the change in PCR bitrate (pcrbitrate=\*) is correctly accounted for  the total TS bitrate (bitrate=\*) starts slowly decreasing even if the actual TS has changed instantly.
```
Jan 02 10:42:18 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26575:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4996100:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425540:duration=7:
Jan 02 10:42:26 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26576:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4996288:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425540:duration=7:
Jan 02 10:42:34 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26626:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=5005688:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425540:duration=8:
Jan 02 10:42:42 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26576:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4996288:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000006:pcrbitrate204=5425539:duration=7:
Jan 02 10:42:50 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26625:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=5005500:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425540:duration=8:
Jan 02 10:42:58 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26576:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4996288:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425539:duration=7:
Jan 02 10:43:06 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=26576:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4996288:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=5000007:pcrbitrate204=5425540:duration=7:
Jan 02 10:43:14 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=25427:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4780276:bitrate=5000007:bitrate204=5425539:userbitrate=5000007:userbitrate204=5425539:pcrbitrate=4773629:pcrbitrate204=5179895:duration=7:
Jan 02 10:43:22 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4931376:bitrate204=5351067:userbitrate=4931376:userbitrate204=5351067:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:43:30 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4902618:bitrate204=5319862:userbitrate=4902618:userbitrate204=5319862:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:43:38 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4855125:bitrate204=5268327:userbitrate=4855125:userbitrate204=5268327:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:43:46 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4817756:bitrate204=5227777:userbitrate=4817756:userbitrate204=5227777:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:43:54 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4801782:bitrate204=5210444:userbitrate=4801782:userbitrate204=5210444:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:44:02 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23979:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4508052:bitrate=4774363:bitrate204=5180691:userbitrate=4774363:userbitrate204=5180691:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:
Jan 02 10:44:10 ts:id=187:services=1:clearservices=1:scrambledservices=0:pids=6:clearpids=6:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=23929:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=4498652:bitrate=4762372:bitrate204=5167680:userbitrate=4762372:userbitrate204=5167680:pcrbitrate=4500006:pcrbitrate204=4882986:duration=7:

```

lelegard, 2020-01-03:
Hi @yaneti,

This is the normal behaviour. Let me explain why.

The TS analysis is performed by a general-purpose C++ class named `TSAnalyzer`. This class is used in the command `tsanalyze`, in the plugin `analyze` and in other circumstances when TSDuck needs to have a global view on the TS. The analyzer can process various types of TS, including incorrect ones. All details are produced and, if some of them are contradictory, it is up to the user to make the synthesis. This is especially true for bitrates. The "TS bitrate" is not some absolute truth in the general case. There are live streams and recorded streams. There are also different ways of avaluating the bitrate and it is up to the user to decide which one is the most appropriate one depending on the context.

For the TS bitrate, there are two sources of information: the "PCR bitrate" and the "user bitrate".

- The PCR bitrate is computed by the analyzer. The result can be correct, approximated or completely wrong (if PCR's are incorrect).
- The user bitrate is some arbitrary value which is externally provided by the "user" (whatever it means). With the command `tsanalyze`, this is the option `--bitrate`. With the plugin `analyze`, this is the bitrate information which is progated from the intput plugin and potentially altered by intermediate plugins (see the `tsp` documentation about bitrate propagation).
- There is a final "TS bitrate" which is equal to the user bitrate of one is specified or to the PCR bitrate if no user bitrate is specified. The user bitrate takes precedence over the PCR bitrate because the user can specify a bitrate which is known to be correct while PCR's can be wrong.

In your case, the TS analysis is restarted every 8 seconds. So, the PCR bitrate is the average PCR bitrate over the last 8 seconds. This is why you see it changing abruptly (with a intermediate value in the interval during which the bitrate change occured).

The user bitrate is the propagated bitrate from `tsp`. If the input plugin is able to report a precise bitrate (demodulator or ASI device for instance), the propagated value will be precise. If the input plugin has no clue about the bitrate (this is the case with UDP), then `tsp` automatically computes an average bitrate from the PCR's. But this is an average _from the beginning of the stream_. This is fine for stable VBR streams. But in the case of a single CBR change, the average slowly changes.

As a conclusion, you need to know your context in order to decide which bitrate value will be the most trustable.

With the plugin `analyze` producing a report at regular intervals:

- If you have a hardware input device which locks on a fixed bitrate, the "user bitrate" is more accurate.
- Otherwise, with a CBR or relatively stable VBR stream, the "user bitrate" and "PCR bitrate" are probably similar.
- With a locally stable input with abrupt changes, the "PCR bitrate" is more accurate over each analysis interval (provided that the PCR values are correct).

yaneti, 2020-01-03:
Hello @lelegard,  Thanks for the explanation.

I still feel that the part "from beginning of the stream" is not expected in the -i context, where I am assuming everything is reset.

What I am actually looking for in the bitrate value is the bitrate in network context, e.g. bits/second no matter if they contain a stream with valid PCR or garbage.  I think from your explanation, since the UDP input plugin can't help me there, my best bet probably would be to take the "bytes"  and use that.

Happy new year :)

closed 2020-01-03 by yaneti

lelegard, 2020-01-03:
> I still feel that the part "from beginning of the stream" is not expected in the -i context, where I am assuming everything is reset.

The bitrate "from beginning of the stream" is what `tsp` is continously evaluating. The "-i context" is local to the `analyze` plugin. These are two independent values. `tsp` is not aware of what each plugin is doing and each plugin is not aware of what the other plugins are doing. Once you understand this, you known what is happening and you can pick up the right value.

> What I am actually looking for in the bitrate value is the bitrate in network context, e.g. bits/second no matter if they contain a stream with valid PCR or garbage. I think from your explanation, [...]

Definitely for your case.