tsduck/tsduck #498: Windows nightly builds with Travis-CI

lars18th, 2020-01-09:
Hi @lelegard ,

The main objective of this issue is to request if you can use the Travis-CI compilation to generate a nigthly build of the Windows portable binaries. It will be very interesting to have direct access to last binaries.

Futhermore, related to the problem of testing the binaries in the Travis-CI environment commented in https://travis-ci.community/t/windows-build-ok-but-exec-fails-because-of-missing-dll/6775/3 I use regulary the windows binaries from the command line without troubles. So I feel the problem is only related to missing internal Windows DLLs. For example, the current `tsduck.dll` declares the use of the `VCRUNTIME140_1.dll` that is missing in my Windows 7 system.

Regards.

lelegard, 2020-01-09:
> The main objective of this issue is to request if you can use the Travis-CI compilation to generate a nigthly build of the Windows portable binaries.

Sorry, I cannot do that for now. A Travis job is limited to 50 mn of execution for open source software (free access). On Linux, the complete build and all tests, utest plus full test suite, take only 10 mn. But, with the Windows platform, the standard TSDuck build is not finished within 50 mn and the job is aborted. So, I had to add parameters to the build procedure to limit the build to a bare minimum to perform some basic tests (which fail anyway because of missing DLL).

This poor performance is probably due to the Windows execution platform at Travis plus the fact that Visual C++ is awfully slow compared to gcc or clang.

The missing DLL issue during testing is due to the fact that Travis uses Windows Server Core edition. On this type of system, there is no GUI and you cannot use and cannot even install anything which is related to graphics, audio and media in general, including DirectShow which is used by TSDuck.

> For example, the current tsduck.dll declares the use of the VCRUNTIME140_1.dll that is missing in my Windows 7 system.

This is different. This DLL is part of Visual C++ runtime. It is expected that the latest version may not be installed everwhere. The installer of TSDuck normally embeds the redistributable version of the VC++ runtime and installs it with TSDuck. Do you encounter problems on Windows 7 after installing TSDuck? I mean a regular installation with the binary installer, not the portable zip which only provides files "as is" without installing anything.

lars18th, 2020-01-09:
> Sorry, I cannot do that for now. A Travis job is limited to 50 mn of execution for open source software (free access). On Linux, the complete build and all tests, utest plus full test suite, take [...]

OK. I hope at some point you can provide it. Perhaps, with a workaround that splits the Windows testing phase in small chunks: compile, testing, packaging, etc. The key is to copy to an external server the binaries compiled by the Travis-CI for the next step.

> The missing DLL issue during testing is due to the fact that Travis uses Windows Server Core edition. On this type of system, there is no GUI and you cannot use and cannot even install anything which is related to graphics, audio and media in general, including DirectShow which is used by TSDuck.

Sorry! I can't understand why TSDuck requires DirectShow/DirectX for command line tools. If you compile it _without_ any capture/play stuff, then it never needs to have such dependencies, rigth?

> This is different. This DLL is part of Visual C++ runtime. It is expected that the latest version may not be installed everwhere. The installer of TSDuck normally embeds the redistributable version [...]

Sorry another time! I'm only using the portable version for Windows and compiled from source for Linux. I never _installed_ a Windows release of TSDuck, nor a Linux deb/rpm package of it.

Regards.

lars18th, 2020-01-10:
Hi @lelegard ,

Regarding your comment in the Travis-CI thread: _"A Windows Server Core environment is not suitable for most Windows applications."_

Please take note that a Windows license is not free. However, the Windows Server Core is part of the Microsoft Hyper-V Server. And this is *free*. So you can install it in any baremetal or virtual machine to run Windows programs. For sure this environment has limitations (for example, it doesn't have support for any Windows Media, DirectX, BDA or similar libraries). So if you want to use travis-CI for Windows, you need to compile TSDuck removing **all** BDA stuff.

I hope it helps.
Regards.

lelegard, 2020-01-13:
Hi @lars18th 

This week-end, I have discovered and tested the recently released "GitHub Actions". This seems amazingly powerful and flexible, much more than Travis-CI. Additionally, this is natively integrated with GitHub, no need to link with external applications such as Travis-CI.

Specifically, there is no restriction on Windows runners. Windows tasks run in the Microsoft Azure cloud and the full TSDuck test suite completes successfully. The fact that GitHub is now a Microsoft-owned company obviously facilitates the potential license issue you mention.

I have setup two tasks. A continuous integration one (on push and on pull request) and a nightly build one (every night around 1 GMT). The CI builds and tests TSDuck on Linux (gcc and clang), macOS and Windows (64 and 32 bits).

The first nightly build completed last night. It builds on Windows only (rebuilding on Linux and macOS from the repo is trivial for users).

Currently, getting the installers back is not really user-friendly but fully functional : Go to https://github.com/tsduck/tsduck/actions (or "Actions" tab in the repo page), click on the latest "Nightly build" and finally click on the "Artifacts" button on the upper right corner.

I plan to automatically upload them to https://tsduck.io/ but I need to wait a few weeks to do that.
- The first solution was to upload directly from the nightly build task. But this requires to store the access credential to the web server in GitHub actions. There is a dedicated "secrets" area for this so that credentials are not exposed to the public. But, working in security for more than 20 years, I am too paranoid to store the credentials of my web server in GitHub, even in a so-called "secrets" area.
- The second solution is to trigger a REST API from the nightly build task on the TSDuck web server. This trigger will make the web server automatically pull the installers from GitHub. This is much better and does not involved any credential storage on either side. However, the PHP code in the web server will need to call the GitHub API to access the installers. But "GitHub Actions" is new and API access is still under development. I have seen in their forum that API access will be available within 6 ou 8 weeks.

So, currently, you have to collect the nightly builds yourself. In a few weeks, when "GitHub Actions" API access is available, the TSDuck web site will collect them and publish them automatically.

lelegard, 2020-01-13:
@lars18th, could you please check that you can access the artifacts of the nightly builds, as indicated in the previous post?

I made a test after logging out from GitHub and all "Actions" part of the repository becomes inaccessible. I would like to make sure that the artifacts are accessible from any logged in user, at least.

lars18th, 2020-01-13:
> @lars18th, could you please check that you can access the artifacts of the nightly builds, as indicated in the previous post?

Yes, I can. Thank you! :+1: 

And I think too that the second solution is the best. However, I suggest to decompress the `installers.zip` file, as it has the four versions (32/64 x EXE/PORTABLE). I'm sure that all users will like only to download one of the options.

> I made a test after logging out from GitHub and all "Actions" part of the repository becomes inaccessible. I would like to make sure that the artifacts are accessible from any logged in user, at least.

Perhaps you want to add too a simple link to "artifacts" in the Wiki: https://github.com/tsduck/tsduck/suites/396810815/artifacts/1084244

lelegard, 2020-01-13:
>  However, I suggest to decompress the installers.zip

Zipping is part of the artifacts upload process. You publish a file (or directory) and the link always points to an archive of it. I cannot change it.

At the next build, the four installers will be published as four separate artifacts. So, there will be four single-file archives.

> Perhaps you want to add too a simple link to "artifacts" in the Wiki: https://github.com/tsduck/tsduck/suites/396810815/artifacts/1084244

I am afraid that this would point to the first nightly builds all the time. And they will disappear in 90 days (artifact retention time).

I did not yet find a way to get a link on the "latest" artifact. This is why we need an API to get it and this API will be used (hopefully) to publish the latest builds on tsduck.io, automatically.

lars18th, 2020-01-13:
> At the next build, the four installers will be published as four separate artifacts. So, there will be four single-file archives.

:tada: 
 
> I am afraid that this would point to the first nightly builds all the time. And they will disappear in 90 days (artifact retention time).
> I did not yet find a way to get a link on the "latest" [...]

You're rigth! :fearful: 
https://help.github.com/en/actions/automating-your-workflow-with-github-actions/persisting-workflow-data-using-artifacts#sharing-data-between-workflow-runs :

- _GitHub does not currently offer a REST API to retrieve uploaded artifacts._

So, no option. But almost link the main page of the nightly builds: https://github.com/tsduck/tsduck/commit/ad424c5a41caf034b4e13d2cf2ca71288dc3cf86/checks?check_suite_id=396810815

Regards.

lelegard, 2020-01-13:
> GitHub does not currently offer a REST API to retrieve uploaded artifacts.

That is what they promised to implement in a few weeks. Let's wait and see.

GitHub Actions is a brand new system. It came up online last month only. But it is so promising that we can wait a bit.

> So, no option. But almost link the main page of the nightly builds: https://github.com/tsduck/tsduck/commit/ad424c5a41caf034b4e13d2cf2ca71288dc3cf86/checks?check_suite_id=396810815

I am not sure if this is a stable URL. Does it point to a specific run of nightly build (the one from last night) or to the nightly builds in general? We will see tomorrow, after the second nighly build if we can find some useful link to the latest build.

lelegard, 2020-01-14:
The second nightly build has completed and I can confirm that the URL above points to a specific build, the first one. I could not find a stable URL on the latest run of nightly builds.

As initially proposed, let's use the manual way first (repo -> actions -> nightly build) and then let GitHub developers work on the Actions API. A better solution will be possible when this API is ready.

lars18th, 2020-01-14:
Hi @lelegard ,

Please, check the packages of the nightly builds. For example, inside the `TSDuck-Win64-3.20-1591-Portable.zip` the directory `TSDuck\setup\` includes the file `vc_redist.arm64.exe` instead of the `vc_redist.x64.exe`.

Regards.

lelegard, 2020-01-14:
Well spotted, thank you, I will fix it.

The naming of the VC++ redistributable library has changed over time (and may change again). To be resilient to name change, the build procedure searches the two wildcards `vc*redist*86.exe` and `vc*redist*64.exe` under `C:\Program Files*\Microsoft Visual Studio`.

I use only Visual Studio Community Edition (the free one) which contains only Intel versions, 32 and 64 bits. I assume that the Azure platforms where GitHub CI runs has the Enterprise Edition with additional architectures. I will just change the wildcard to `vc*redist*x64.exe` to avoid arm.

lelegard, 2020-01-31:
GitHub has released a beta of the [Actions API](https://developer.github.com/v3/actions/).

At the end of the nightly build, after uploading the installers ("artifacts" as they say in GitHub Actions), the workflow triggers a PHP script on tsduck.io which, in turn, uses the Actions API to retrieve the installers and moves them automatically in the [prerelease area](https://tsduck.io/download/prerelease/). I used this area to push some builds from time to time. Now, it will be automatically updated every night.

Because the Actions API is beta and subject to change, some disruptions may occur in the near future. I will fix them when I have time.

lars18th, 2020-02-01:
Hi @lelegard ,

Thank you for this valuable functionality! And thank you to the people of Github for the Actions API.

However, one question: These "nightly builds" will replace the "pre-releases"?

lelegard, 2020-02-01:
> one question: These "nightly builds" will replace the "pre-releases"?

Well, it depends on the meaning you want to place behind words. For me, "pre-releases" are just builds which are not yet official releases. So, nightly builds are automatically provided pre-releases. I no longer need to manually push Windows pre-releases. If necessary, I may push manually built pre-releases for Linux.

lars18th, 2020-02-01:
OK, then these "automatically" generated builds at every night will replace the current "manual" pre-releases. That's fine for me.

So, if you agree then I close this issue.
Thank you for the enhancement!

closed 2020-02-01 by lelegard

lars18th, 2020-02-05:
Hi @lelegard ,

Perhaps a small aesthetic request: Can do you add the "date" of the building in the nightly builds?
I suggest two options:

1. Write the date in the title: `Pre-release version 3.20-1651 created 2020/02/04`
2. Insert the date in the filename: `TSDuck-Win32-3.20-1651-20200204-Portable.zip`

Any of the two are valid.
Regads.