tsduck/tsduck #503: Request: Function to manage PCR and OPCR

lars18th, 2020-01-14:
Hi @lelegard ,

What's the best method to change PCR and OPCR values? For example, I want to _copy_ the PCR inside the OPCR, or _interchange_ the values. You have considered this?

I await your response.
And thank you for this project!

lelegard, 2020-01-14:
Programmatically, the accessor methods exist in class TSPacket. You can safely modify an existing PCR or OPCR value. If the value did not previously exist and there is no or not enough stuffing in the adaptation field, creating space for PCR or OPCR will truncate the payload.

lars18th, 2020-01-14:
> Programmatically, the accessor methods exist in class TSPacket. You can safely modify an existing PCR or OPCR value.

The idea is to provide a plugin to do this task.

> If the value did not previously exist and there is no or not enough stuffing in the adaptation field, creating space for PCR or OPCR will truncate the payload.

Perhaps a new plugin to "expand" stuffing in the TS Header will help. It can be done using the current model of _packet processing_ storing the overflow of the payload in a _virtual_ TS packet for each pid. Then when the PES ends, you can use a null packet to write the content of this virtual packet. What you think? Is it a waste of time or is it useful?