tsduck/tsduck #504: es -> pes -> ts

dibbitson, 2020-01-14:
Hello,

I'm exploring using TSDuck (via c++ library) to take elementary streams (x264 and aac bitstreams) and create a transport stream from them, which will subsequently be sent over a network via TCP.

Is this possible using TSDuck? I'm not yet familiar enough with all the concepts of TS to tell from the documentation if it is.

Any help getting started would be much appreciated.

Thanks!

lelegard, 2020-01-14:
For what you are intending to do, you need a MUX and, despite repeated users requests, TSDuck is not a MUX. To mux audio and video and getting a TS, FFmpeg (and its libav library) are more appropriate.

Then, if you need more complex manipulations on the TS, TSDuck is the right tool. 

By experience, users are often mixing FFmpeg and TSDuck because they are complementary toolsets. FFmpeg takes care of audio / video processing and muxing. TSDuck takes care of the transport stream system manipulation and routing and transmission.

dibbitson, 2020-01-14:
Thanks for the response, this clears some questions up. Cheers.

closed 2020-01-14 by dibbitson