tsduck/tsduck #506: Help needed to investigate SRT input flow control problem on Windows

lelegard, 2020-01-19:
Dear TSDuck users,

### About SRT

Thanks to a very good code contribution from @spnngl (Anthony Delannoy), TSDuck is now able to input and output using [SRT](https://www.srtalliance.org/) (Secure Reliable Transport) in the form and an input and output plugin named `srt` (pull requests #500 and #505).

The availability of this plugin depends on the presence of [libsrt](https://github.com/Haivision/srt/) during compilation of TSDuck. This open source library is now available on Fedora 31, Ubuntu 19.04 and macOS. Additional distros will follow.

If libsrt is not available, building TSDuck completes successfully but using the `srt` plugin fails with error _"This version of TSDuck was compiled without SRT support"_. So, this is pretty straightforward to build, deploy and use.

On Windows, as often with open source libraries, there is no pre-built binaries for libsrt and building it is a pain (no precise intructions and fend for yourself). To automate the build, I have made the effort to create a separate project named [srt-win-installer](https://github.com/tsduck/srt-win-installer) which contains scripts to rebuild librt on Windows and creates a binary installer. The latest binary installers for libsrt on Windows are available in the [release area](https://github.com/tsduck/srt-win-installer/releases) of the project.

Building TSDuck on Windows adopts the same strategy as on Linux and macOS: if the binary for libsrt has been installed, it is automatically used. Otherwise, the same _"compiled without SRT support"_ error is reported.

The libsrt is linked as static library. So, deploying TSDuck on Windows with SRT support does not need any additional DLL. The [building prerequisites](https://tsduck.io/doxy/building.html) have been updated accordingly.

### The problem about Windows

That's for the good news. Now, the not-so-good news: It seems that the _input_ `srt` plugin does not work very well on Windows (it's OK on Linux and macOS). See the details below.

I am not an expert in SRT (actually meaning "completely ignorant") and I have no time and little motivation to become one. Anthony, the author of the `srt` plugins, does not have any Windows system and does not seem to like it (I can't blame him). So, if anyone is interested in SRT support on Windows, some volunteers have to diagnose and fix that issue.

Please comment on the problem using this thread and submit a [pull request](https://tsduck.io/doxy/contributing.html) for any fix.

### The details of the tests

In each test, a sender system loops on an existing TS file and sends it using SRT, acting as "listener", waiting for a receiver system to connect. A receiver system connects to the sender and receives the TS using SRT. After 30 seconds, the receiver stops and displays a summary of the TS analysis.

In my configuration, I have a Mac system with VMware and several virtual machines (Windows 10, Fedora, Ubuntu, CentOS, Alpine Linux). I made tests between the Mac host, the Windows VM and the Fedora VM, all of them using libsrt version 1.4.1.

Here is the sender -> receiver success / failure matrix:

- Linux -> macOS : OK
- macOS -> Linux : OK
- Windows -> macOS : OK
- macOS -> Windows : errors, apparently due to control flow failures

In all cases, on the sender system (listener), with `$input` being a TS file, the command is:
~~~
tsp -v -I file $input --infinite -P regulate -O srt 12345 --transtype live --messageapi
~~~

On the receiver system, with `$sender` being the host name or address of the sender system:
~~~
tsp -v -I srt $sender:12345 --transtype live --messageapi --conn-timeout 100000 -P until --seconds 30 -P analyze --ts -O drop
~~~

In the failing case (macOS -> Windows), both parties continuously display errors.

On macOS sender:
~~~
16:02:27.724840*E:SRT.d: SND-DROPPED 3 packets - lost delaying for 1021ms
16:02:27.725196*E:SRT.d: SND-DROPPED 21 packets - lost delaying for 1021ms
16:02:27.734991*E:SRT.d: SND-DROPPED 55 packets - lost delaying for 1021ms
16:02:27.745282*E:SRT.d: SND-DROPPED 48 packets - lost delaying for 1021ms
...
~~~

On Windows 10 receiver:
~~~
16:02:28.512000*E:SRT.c: %20021859:No room to store incoming packet: offset=4751 avail=1714 ack.seq=782802684 pkt.seq=782807435 rcv-remain=6477
16:02:28.512000*E:SRT.c: %20021859:No room to store incoming packet: offset=4752 avail=1715 ack.seq=782802684 pkt.seq=782807436 rcv-remain=6476
16:02:28.521000*E:SRT.c: %20021859:No room to store incoming packet: offset=4797 avail=1727 ack.seq=782802684 pkt.seq=782807481 rcv-remain=6464
16:02:28.521000*E:SRT.c: %20021859:No room to store incoming packet: offset=4804 avail=1728 ack.seq=782802684 pkt.seq=782807488 rcv-remain=6462
...
~~~

But Windows receives enough TS packets to make a valid analysis of the stream. So, the problem is a loss of packets in an otherwise working transmission. This looks like a flaw in the flow control.

As mentioned at the beginning of this message, I will not investigate this problem further. I let competent and motivated people in SRT proceed.

For initial testing of TSDuck with SRT on Windows, you may use the latest installers from the nightly builds. To access these nightly builds, follow these steps:
- Login to GutHub (anonymous access not allowed)
- Go to the "Actions" tab of the TSDuck repo (https://github.com/tsduck/tsduck/actions)
- Click on the top-most "Nightly build"
- Click on "Artifacts (5)" in the upper-right corner.
- Select your favorite package (installer / portable, 32 / 64 bits)

If you need to debug the issue, you may need to [rebuild TSDuck on Windows](https://tsduck.io/doxy/building.html).

maxsharabayko, 2020-01-20:
@lelegard  @spnngl Thank you for bringing support for SRT into TSDuck!

> No room to store incoming packet

This error message means that a packet has arrived, but there is no room in the receiver's buffer to store it.
The receiver's buffer is 90 Mbits by default, and can not be increased in the runtime. By default, the buffer holds at least 120 ms of data (according to SRTO_LATENCY value).
The sise of the buffer can be set to a bigger value with the corresponding socket option `SRTO_RCVBUF`.

For the test described above to be fully reproducible, could you please share the TS file used for testing. Or at least the characteristics of the source file used.

lelegard, 2020-01-20:
Hi @maxsharabayko,

The file is here: https://tsduck.io/streams/france-astra-19.2E/astra192E-ts1080-2018-05-11.ts

This is part of the TSDuck sample stream repository here: https://tsduck.io/streams/

maxsharabayko, 2020-01-21:
Hi @lelegard  There was a change in internal timing in SRT after v1.4.1. 
Would you be so kind to check if this issue is still valid with the latest SRT master? Only the macOS -> Windows case.

lelegard, 2020-01-21:
Hi @maxsharabayko,

It is true that the version of libsrt on Windows was not exactly 1.4.1 since it was built from the latest master at that time. So, it was _"some commits after 1.4.1"_. Linux and macOS, on the other hand, have (presumably) version 1.4.1 exactly because they were installed from the official repos of those distros (homebrew for macOS).

I have updated the scripts to fetch a more accurate version based on `git describe`. After a rebuild (and an even more recent master branch), the version is now 1.4.1-57-47b2553, meaning commit 47b2553, 57 commits after v1.4.1. See the [release area](https://github.com/tsduck/srt-win-installer/releases).

So, I did the same tests, both ways, and the result is ... worse.

- In the direction macOS -> Windows, I have the same error messages as previously.
- In the direction Windows -> macOS, it no longer works. I start the sender first which waits for an incoming connection. Then, I start the receiver on macOS which connects to the sender on the Windows VM. At that time, the sender aborts right after receiving the connection with the following message:
~~~
* Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters
~~~
The prefix comes from TSDuck, the message _"Operation not supported: Bad parameters"_ comes from `srt_getlasterror_str()`.

maxsharabayko, 2020-01-24:
Did some quick connection and transmission tests with [xtransmit](https://github.com/maxsharabayko/srt-xtransmit). Didn't see these issues so far.
I was thinking if the macOS -> Windows has something to do with the wrong streaming rate from TS?
That would explain the message:
```
No room to store incoming packet
```
Still not clear why macOS -> macOS workds then.

lelegard, 2020-01-24:
Hi @maxsharabayko (and @spnngl for your info),

I have rebuilt libsrt Windows binaries from the latest repo (commit https://github.com/Haivision/srt/commit/c59d413a0e1ee8486deb61d65e0134b4892d022a), which is 10 commits ahead from my latest build. The libsrt installer is now version 1.4.1-67-c59d413. See the [release area](https://github.com/tsduck/srt-win-installer/releases).

I no longer have problems when Windows is the sender system. However, there are still errors when Windows is the receiver. The sender can be Linux or macOS, the result is the same. More details below.

The configuration is the following:
- Host: macOS 10.15.2, name "hostmac"
- VM: Fedora Linux 31, name "vmfedora"
- VM: Windows 10 version 1909, name "vmwindows"

TSDuck was rebuilt with libsrt version 1.4.1 on macOS and Fedora distros and fresh version 1.4.1-67-c59d413 on Windows.

On hostmac:
~~~
$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.20-1627
Built Jan 24 2020 - 20:16:23
Using Clang 11.0.0 (clang-1100.0.33.17), C++ std 2011.03
Web library: libcurl: 7.64.1, ssl: SecureTransport (LibreSSL/2.8.3), libz: 1.2.11
SRT library: libsrt version 1.4.1
Dektec: This version of TSDuck was compiled without Dektec support
~~~
On vmfedora:
~~~
$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.20-1627
Built Jan 24 2020 - 22:02:48
Using GCC 9.2.1, C++ std 2011.03
Web library: libcurl: 7.66.0, ssl: OpenSSL/1.1.1d-fips, libz: 1.2.11
SRT library: libsrt version 1.4.1
Dektec: DTAPI: 5.37.0.134
~~~
On vmwindows:
~~~
PS Q:\test> tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.20-1627
Built Jan 24 2020 - 21:27:29
Using MSVC 19.24.28315.00, C++ std 1997.11
Web library: WinInet
SRT library: libsrt version 1.4.1-67-c59d413
Dektec: DTAPI: 5.37.0.134, Service: 3.7.0.83
~~~

Sender command:
~~~
tsp -v -I file $input --infinite -P regulate -O srt 12345 --transtype live --messageapi
~~~
The `$input` file is available at https://tsduck.io/streams/france-astra-19.2E/astra192E-ts1080-2018-05-11.ts

Receiver command (`$sender` being one of hostmac, vmfedora, vmwindows):
~~~
tsp -v -I srt $sender:12345 --transtype live --messageapi --conn-timeout 100000 -P until --seconds 30 -P analyze --ts -O drop
~~~

Result matrix:
- Linux -> macOS : OK
- macOS -> Linux : OK
- Windows -> Linux : OK
- Windows -> macOS : OK
- Linux -> Windows : errors
- macOS -> Windows : errors

Linux or macOS sender (hundredths of similar errors):
~~~
23:36:00.045663/tsp*E:SRT.d: SND-DROPPED 107 packets - lost delaying for 1030ms
23:36:00.055149/tsp*E:SRT.d: SND-DROPPED 1 packets - lost delaying for 1021ms
~~~
Windows receiver (hundredths of similar errors):
~~~
23:36:00.043000*E:SRT.c: @19579477:No room to store incoming packet: offset=7022 avail=6512 ack.seq=402157218 pkt.seq=402164240 rcv-remain=1679
23:36:00.044000*E:SRT.c: @19579477:No room to store incoming packet: offset=7023 avail=6513 ack.seq=402157218 pkt.seq=402164241 rcv-remain=1678
~~~

The stream is a complete DVB-S2 transport stream. It is regulated at approximately 57.49 Mb/s. Assuming that the rate was maybe too high, I extracted an SPTS (one single service) using the command below. The stream is then regulated at approximately 5.97 Mb/s. But the same errors are present. Visually, there are less errors, they appear not as fast as previously. But there are 10 times less packets.

~~~
tsp -v -I file $input --infinite -P zap canal+ -P pcrbitrate -P regulate -O srt 12345 --transtype live --messageapi
~~~

lelegard, 2020-01-24:
One more info: the network between hostmac, vmfedora and vmwindows is the VMware "host only" virtual network.

lelegard, 2020-01-25:
Hi @maxsharabayko,

Just to make sure that VMware and virtual networks are not involved in the cause of the problem, I made the same test between a physical Windows laptop (same Windows 1909) and the physical Mac using Wifi network. The result is exactly the same, same error messages on the sender (Mac) and receiver (Windows), maybe a bit less messages.

closed 2025-02-24 by lelegard