tsduck/tsduck #507: Overloaded function issue

spnngl, 2020-01-20:
Hi,

Few weeks ago I made an application following the sample-app from TSDuck, but since commit 0c66d4dc20f881a17bc8316f697e0e77f333e768 I have an issue with the ts::Args::analyze function(s).

#### Code sample
```
ts::TSPacket packetBuffer[BUFSIZE];
ts::TSPacketMetadata packetMetadata[BUFSIZE];
  
ts::NewInputProfile inputPluginAllocator = ts::PluginRepository::Instance()->getInput(u"ip", tsp);
if (inputPluginAllocator == nullptr)
    return (-1);
  
ts::SafePtr<ts::InputPlugin> inputPlugin(inputPluginAllocator(&tsp));
if (inputPlugin.isNull())
    return (-1);
  
// Create instance of output plugin
ts::NewOutputProfile outputPluginAllocator = ts::PluginRepository::Instance()->getOutput(u"drop", tsp);
if (outputPluginAllocator == nullptr)
    return (-1);
  
ts::SafePtr<ts::OutputPlugin> outputPlugin(outputPluginAllocator(&tsp));
if (outputPlugin.isNull())
    return (-1);

if (!inputPlugin->analyze(u"ip", u"239.255.12.1:2000"))
    return (-1);
  
if (!(inputPlugin->getOptions() && inputPlugin->start()))
    return (-1);
```

Before this commit the `bool ts::Args::analyze(const UString& app_name, const UStringVector& arguments, bool processRedirections)` was called. After this commit and for the same code the `bool ts::Args::analyze(int argc, char* argv[], bool processRedirections)` is called.

#### Backtraces

* before:
```
Breakpoint 1, ts::Args::analyze (this=0x7fffec9ceae8, processRedirections=true) at tsArgs.cpp:769
769     for (IOptionMap::iterator it = _iopts.begin(); it != _iopts.end(); ++it) {
(gdb) bt
#0  ts::Args::analyze (this=0x7fffec9ceae8, processRedirections=true) at tsArgs.cpp:769
#1  0x00007ffff6642c66 in ts::Args::analyze (this=0x7fffec9ceae8, app_name=..., arguments=std::vector of length 1, capacity 1 = {...}, processRedirections=true) at tsArgs.cpp:746
#2  0x000000000055854d in main (argc=3, argv=0x7fffffffd968, envp=0x7fffffffd988) at ../duck/cc/main.cc:78
```

* after:
```
Breakpoint 1, ts::Args::analyzeInternal (this=0x7fffec9cfae8, processRedirections=true) at tsArgs.cpp:797
797     for (IOptionMap::iterator it = _iopts.begin(); it != _iopts.end(); ++it) {
(gdb) bt
#0  ts::Args::analyzeInternal (this=0x7fffec9cfae8, processRedirections=true) at tsArgs.cpp:797
#1  0x00007ffff6642651 in ts::Args::analyze (this=0x7fffec9cfae8, argc=-22096, argv=0x7fffffffa810, processRedirections=true) at tsArgs.cpp:786
#2  0x000000000055854d in main (argc=3, argv=0x7fffffffd968, envp=0x7fffffffd988) at ../duck/cc/main.cc:78
```

I do not understand why this happen, hope someone knows why

spnngl, 2020-01-20:
Yes sorry forgot the curly braces when writing on github the IP address.
My code uses UStringVector and add an interface if specified.

Here the code I replaced by the IP address in the example above to simplify.
```
 71     ts::UStringVector ip_vector;
 72     ip_vector.push_back(mc_ip);
 73     if (interface.size() > 2) {
 74         ip_vector.push_back(u"-l");
 75         ip_vector.push_back(interface);
 76     }
```

lelegard, 2020-01-24:
Hi @spnngl,

The new API in class `ts::TSProcessor` will simplify your code a lot. See `tsp.cpp` and the sample application in `sample/sample-app`.

spnngl, 2020-01-27:
Hi @lelegard,

Thanks I will look into that

spnngl, 2020-01-28:
I made the modifications and it works better. But I think there is a memory leak somewhere, 29edc7abe2ae95b935488efa9225404b8aced7f6 commit, current master.

Tested with the command ```tsp -I ip ... -P count -O drop```

I'll try to find the source.

closed 2020-01-28 by spnngl