tsduck/tsduck #511: PORTABLE script

lars18th, 2020-01-24:
Hi @lelegard ,

I suggest to provide some script to create (in fact, to copy all binaries from a compilation) to a destination _PORTABLE_ subdirectory. With this any installation will not be required, and the execution of the scripts `setenv.sh` neither (`source /root/tsduck/src/tstools/release-x86_64/setenv.sh`).

I do one initial example `create-portable.sh`:
```bash
#!/bin/bash
#
# Copy all stuff to a target directory as a PORTABLE release.

BASEDIR_LIBS=./src/libtsduck/release-x86_64/
BASEDIR_TOOL=./src/tstools/release-x86_64/
BASEDIR_PLUG=./src/tsplugins/release-x86_64/

TARGET=$1

# Copy Library stuff
cp -t $TARGET $BASEDIR_LIBS/libtsduck.a
cp -t $TARGET $BASEDIR_LIBS/tsduck.so

# Copy Tools
find $BASEDIR_TOOL/ ! -name "*.*" | xargs cp -t $TARGET

# Copy Plugins
cp -t $TARGET $BASEDIR_PLUG/*.so
```

I hope you want to provide one.
Regards.

lars18th, 2020-01-24:
Hi,

From the point of view of a dockerized installation or to testing different versions it will be very useful to have an _isolated_ binary directory with all the stuff and no dependencies. :wink:

lelegard, 2020-01-24:
It already exists in this form:
~~~
make install SYSPREFIX=$TARGET
~~~

This is explained [here](https://tsduck.io/doxy/building.html#nonstdinst).

lars18th, 2020-01-24:
> It already exists in this form:
> 
> ```
> make install SYSPREFIX=$TARGET
> ```
> 
> This is explained [here](https://tsduck.io/doxy/building.html#nonstdinst).

Sorry, I don't agree.

First of all, this script doesn't work if the passed parameter has `/` at end.
Futhermore, it executes a lot of other commands, like installing security rules not required in a PORTABLE install.
And, I feel (I'm not sure) it cleans some files from the compilation directory. And then any new recompilation requires more work.

So, why not support something like `make portable TARGET` ?
You agree?

lelegard, 2020-01-24:
> First of all, this script doesn't work if the passed parameter has / at end.

This is not a script, this is a `make` command which follows the `make` syntax. It makes the target `install` and sets the variable `SYSPREFIX` to the target root directory of your portable installation (it must be an absolute path because `make` recurses). 

Any other "parameter" you would pass would be interpreted as an additional make target.

> Futhermore, it executes a lot of other commands, like installing security rules not required in a PORTABLE install.

This command rebuilds binaries which need to be rebuilt (which is a required) and installs them in `$TARGET/bin`. The only thing that is not completely necessary in that case is the copy of 2 (only 2) system configuration text files under `$TARGET/etc`. This is not necessary but it does not hurt either. If if it makes you really sick, just `rm -rf $TARGET/etc` afterwards.

So, this command is a strict definition of a portable installation. A zip archive in the Windows way. A `make install` command is the traditional Unix way.

> And, I feel (I'm not sure) it cleans some files from the compilation directory. And then any new recompilation requires more work.

It does not do anything in addition, no cleanup, no unnecessary build.

> So, why not support something like make portable TARGET ?

Because this is not a valid `make` command...

lelegard, 2020-01-24:
I have to add that "portable" is a typical Windows vocabulary. It was introduced by frustrated non-admin users after Windows NT because of the unnecessarily complex and heterogeneous installation processes on Windows.

In the Unix world, self-sufficient applications in their own environment have been the rule from the beginning.

In the Unix world (in fact, in the whole computing world except maybe on Windows), "portable" means "which can be compiled on various architectures, OS and compilers". Which is a very different concept....

As a long-time Unix user (I started using it in 1984), I have always understood the word "portable" in the latter meaning.

closed 2020-01-31 by lelegard