tsduck/tsduck #512: Last build fails with gcc-4.9

lars18th, 2020-01-24:
Hi @lelegard ,

Current repo fails to compile with the GCC-4.9 backport:
```
make CC=/usr/lib/gcc-4.9-backport/bin/gcc GCC=/usr/lib/gcc-4.9-backport/bin/gcc CXX=/usr/lib/gcc-4.9-backport/bin/g++ AR=/usr/lib/gcc-4.9-backport/bin/gcc-ar ARFLAGS=rc NOTEST=1 NOPCSC=1 NODTAPI=1

[...]
  [CXX] dtv/descriptors/tsAbstractDeliverySystemDescriptor.cpp
In file included from /home/user/tsduck/src/libtsduck/dtv/tsModulation.h:37:0,
                 from /home/user/tsduck/src/libtsduck/dtv/tsModulationArgs.h:39,
                 from dtv/descriptors/tsAbstractDeliverySystemDescriptor.h:37,
                 from dtv/descriptors/tsAbstractDeliverySystemDescriptor.cpp:30:
/home/user/tsduck/src/libtsduck/dtv/tsDeliverySystem.h:53:28: error: '::SYS_DVBC_ANNEX_A' has not been declared
         DS_DVB_C_ANNEX_A = ::SYS_DVBC_ANNEX_A,
                            ^
/home/user/tsduck/src/libtsduck/dtv/tsDeliverySystem.h:55:28: error: '::SYS_DVBC_ANNEX_C' has not been declared
         DS_DVB_C_ANNEX_C = ::SYS_DVBC_ANNEX_C,
                            ^
/home/user/tsduck/src/libtsduck/dtv/tsDeliverySystem.h:63:28: error: '::SYS_DTMB' has not been declared
         DS_DTMB          = ::SYS_DTMB,
                            ^
In file included from /home/user/tsduck/src/libtsduck/dtv/tsModulationArgs.h:39:0,
                 from dtv/descriptors/tsAbstractDeliverySystemDescriptor.h:37,
                 from dtv/descriptors/tsAbstractDeliverySystemDescriptor.cpp:30:
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:111:20: error: '::QAM_4_NR' has not been declared
         QAM_4_NR = ::QAM_4_NR,
                    ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:200:20: error: '::FEC_2_5' has not been declared
         FEC_2_5  = ::FEC_2_5,
                    ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:399:20: error: '::TRANSMISSION_MODE_C1' has not been declared
         TM_C1    = ::TRANSMISSION_MODE_C1,
                    ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:400:20: error: '::TRANSMISSION_MODE_C3780' has not been declared
         TM_C3780 = ::TRANSMISSION_MODE_C3780,
                    ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:446:24: error: '::GUARD_INTERVAL_PN420' has not been declared
         GUARD_PN420  = ::GUARD_INTERVAL_PN420,
                        ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:447:24: error: '::GUARD_INTERVAL_PN595' has not been declared
         GUARD_PN595  = ::GUARD_INTERVAL_PN595,
                        ^
/home/user/tsduck/src/libtsduck/dtv/tsModulation.h:448:24: error: '::GUARD_INTERVAL_PN945' has not been declared
         GUARD_PN945  = ::GUARD_INTERVAL_PN945,
                        ^
make[2]: *** [release-x86_64/tsAbstractDeliverySystemDescriptor.o] Error 1
make[1]: *** [default] Error 2
make: *** [default] Error 2
```

Please, can you fix it?
Thank you!

lelegard, 2020-01-24:
I do not think that this is a compiler version issue but a kernel version one. Which kernel are you using?

The undefined symbols come from `/usr/include/linux/dvb/frontend.h`, a system header which defines the DVB API. It comes with the kernel.

The current versions are kernel 5.4.14 and gcc 9.2. The oldest Linux platform on which I continue to build TSDuck is CentOS 7.7 which comes with kernel 3.10.0 and gcc 4.8.5. And it builds and works. These symbols were already defined in kernel 3.10, a rather old one. Yours must be even older.

I cannot support old kernels which have been obsolete for a long time. TSDuck already supports three compilers (GCC, LLVM, MSVC) in various versions, multiple OS, including Windows, macOS, several Linux distros (Red-Hat family: Fedora / CentOS / RHEL, Debian family: Ubuntu / Debian / Raspbian, Alpine and its exotic libc) with various combinations of compilers, kernels, libraries and system configuration. I cannot do more. Maybe some old version of TSDuck will build on some old version of Linux, but this will be just old crap...

lars18th, 2020-01-24:
Hi,

I'm using `3.2.0-4-amd64` (Debian 7 64bit). And I really want to be compatible with this kernel. Please, think that previous versions are still compiling fine with 4.9-backport. So if the problem is a simple definition or declaration, then it will be easy to ovecome it.

For example:
```
# cat /usr/include/linux/dvb/frontend.h | grep SYS_DVB
        SYS_DVBC_ANNEX_AC,
        SYS_DVBC_ANNEX_B,
        SYS_DVBT,
        SYS_DVBS,
        SYS_DVBS2,
        SYS_DVBH,
        SYS_DVBT2,
```

Then the problem is an old frontend values. But the TSDuck is in fact *NOT* using the kernel for this pourpose, as I don't use any DVB Tuner on it. Futhermore, some platforms (like Android, Raspberry, etc.) can be using older kernel headers.

Please, can you provide some workaround? I repeat it: versions from two weeks ago compile fine.
Thank you!

lelegard, 2020-01-24:
> I repeat it: versions from two weeks ago compile fine.

Not on the same kernel, definitely impossible. Those symbols have been used for months. They were already used when TSDuck version 3.19 was released.

lelegard, 2020-01-24:
Current Debian is 10. Debian 7 was released in 2013, 7 years ago. Even as a "Long-Term Support" version, it is no longer supported by Debian since 2018. I see no reason why a free unpaid single-developer project such as TSDuck should support it in 2020.

lars18th, 2020-01-24:
Hi, I'll do more tests... I can't understand the reason to be broken now. Some weeks ago it compiles without troubles.

lelegard, 2020-01-31:
Fixed with PR #513

closed 2020-01-31 by lelegard