tsduck/tsduck #514: Continue running even if one or more channels is unavailable

morpig, 2020-01-26:
Hi there,

I'm currently experiencing an issue where I'm using DVB-T2 for input, and output to HLS for multiple channels by using `-P fork` for every channels. 

However, if one of the channels in that mux went disappear/missing from the transport stream, is there anyway to not exit and continue run the program? Currently, it will spit out `* Error: zap: service id 0x0005 not found in PAT` and exit the program.

Is there anyway to ignore and keep continue running and if the channel is back, it'll process the output?

Thanks! Sorry for the bad grammars.

lelegard, 2020-01-26:
Currently, the plugin `zap` aborts when the target channel is missing. A new option will be added to continue passing an empty stream until the service is back.

In the meantime, if you do not already use it, you may use `-P fork --ignore-abort` so that the terminated process does not cause the parent `tsp` to abort as well. At least, the other service will continue.

lars18th, 2020-01-31:
> Currently, the plugin `zap` aborts when the target channel is missing. A new option will be added to continue passing an empty stream until the service is back.

One suggestion for this _new_ option for the `zap` pluging. Please, add support to _start_ even if the program doesn't exist. Then it will possible to catch all cases: found, not found, removed and restored.

Thank you for this great tool!

lelegard, 2020-01-31:
> Please, add support to start even if the program doesn't exist. 

The idea is that a service may come and go in a TS. There is no reason to assume that the service is up when we start. So, yes this is planned. I just need to find some time. It is more complicated than it looks because of the combination of possible situations.

lars18th, 2020-02-01:
> I just need to find some time. It is more complicated than it looks because of the combination of possible situations.

Don't worry. It's fine. Any improvements you make are welcome and much appreciated.

lelegard, 2020-02-28:
The option `--ignore-absent` has been added to plugin `zap`.

lars18th, 2020-02-28:
> The option `--ignore-absent` has been added to plugin `zap`.

Great! Thank you for this improvement. :+1:

closed 2020-03-04 by lelegard