tsduck/tsduck #515: Does the inject plugin reverse the section order ?

bobben1no, 2020-01-26:
Hi,

I have been playing around with the inject plugin with works great but I notice the NIT sections seems to be reversed ?   I am using tstables to extract binary sections which is decompiled to XML by using tstabcomp .  I can see that a complete NIT is collected and stored in the XML file(but not the individual sections). All XML data except NIT is removed from the XML file. 

(I have been using 3.19-1520 and 3.19-1411 in Ubuntu 19.10)

The section order I get when injecting this NIT.xml SI file is :
section 8 of 8
section 7 of 8
section 6 of 8
section 5 of 8
section 4 of 8
section 3 of 8
section 2 of 8
section 1 of 8
section 0 of 8

I assume it is the inject plugin that compiles the sections in this order. 

ETSI EN 300 468 specifies :
`section_number: This 8-bit field gives the number of the section. The section_number of the first section in thesub_table shall be "0x00". The section_number shall be incremented by 1 with each additional section with the same table_id and network_id`

lelegard, 2020-01-26:
Hi @bobben1no,

The sections are injected in sequential order, not in reverse order as you mention. I can show this on a simple example.

File `nit.xml`:
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT network_id="1">
    <transport_stream transport_stream_id="10" original_network_id="1234" preferred_section="0"/>
    <transport_stream transport_stream_id="11" original_network_id="1234" preferred_section="1"/>
    <transport_stream transport_stream_id="12" original_network_id="1234" preferred_section="2"/>
    <transport_stream transport_stream_id="13" original_network_id="1234" preferred_section="3"/>
    <transport_stream transport_stream_id="14" original_network_id="1234" preferred_section="4"/>
    <transport_stream transport_stream_id="15" original_network_id="1234" preferred_section="5"/>
    <transport_stream transport_stream_id="16" original_network_id="1234" preferred_section="6"/>
    <transport_stream transport_stream_id="17" original_network_id="1234" preferred_section="7"/>
    <transport_stream transport_stream_id="18" original_network_id="1234" preferred_section="8"/>
  </NIT>
</tsduck>
~~~

Command and its output:
~~~
$ tsp -I file input.ts -P inject nit.xml --replace --pid 16 -P tables -p 16 -a -O drop | grep Section:
  Section: 0 (last: 8), version: 0, size: 22 bytes
  Section: 1 (last: 8), version: 0, size: 22 bytes
  Section: 2 (last: 8), version: 0, size: 22 bytes
  Section: 3 (last: 8), version: 0, size: 22 bytes
  Section: 4 (last: 8), version: 0, size: 22 bytes
  Section: 5 (last: 8), version: 0, size: 22 bytes
  Section: 6 (last: 8), version: 0, size: 22 bytes
  Section: 7 (last: 8), version: 0, size: 22 bytes
  Section: 8 (last: 8), version: 0, size: 22 bytes
  Section: 0 (last: 8), version: 0, size: 22 bytes
  Section: 1 (last: 8), version: 0, size: 22 bytes
  Section: 2 (last: 8), version: 0, size: 22 bytes
  Section: 3 (last: 8), version: 0, size: 22 bytes
  Section: 4 (last: 8), version: 0, size: 22 bytes
  Section: 5 (last: 8), version: 0, size: 22 bytes
  Section: 6 (last: 8), version: 0, size: 22 bytes
  Section: 7 (last: 8), version: 0, size: 22 bytes
  Section: 8 (last: 8), version: 0, size: 22 bytes
  Section: 0 (last: 8), version: 0, size: 22 bytes
  Section: 1 (last: 8), version: 0, size: 22 bytes
  Section: 2 (last: 8), version: 0, size: 22 bytes
  Section: 3 (last: 8), version: 0, size: 22 bytes
  ...
~~~

How do you determine that the section order is reversed?

Concerning what ETSI EN 300 468 says, it does not mean that the sections shall be injected in sequential order. The "first" section means the first logical section. When a receiver gets sections, each section contains its own index and last index. So, when the first section is received (first to be received, not always section index 0), the receiver knows the total number of sections. As sections arrive, they are stored in their respective index, until all sections are there. This is exactly how TSDuck behaves. 

That being said, it is very common that sections are inserted in sequential order and this is what TSDuck does as well. But, even with a sequential order, receivers catch section on non-sequential order because sections are cycled. When you tune to a transponder, there is no reason to jump in the stream just before section 0. So, assuming that right after you tune on a frequency, the next section is number 5, then the NIT sections will be read in order 5 6 7 8 0 1 2 3 4. At section 4, the NIT is complete and can be processed.

So, while inserting sections in sequential order is very common, it is neither mandated by the standard nor a requirement for proper functioning.

bobben1no, 2020-01-26:
Your example work as expected. 

I think my problem is caused by a combination of  looping ts file  `tsp -I file --infinite input.ts` with repetition rate. When I remove the rate parameter the section order is "normal".  I want to create a continuous NIT feed with control over the repetition rate and I probably do something wrong.

This gives normal order but no control over repetition rate:
`tsp -I file input.ts -P inject nit.xml --replace --pid 16 -P tables -p 16 -a -O drop | grep Section:` 

This gives reversed section order and repetition rate is not correct:
`tsp -I file --infinite input.ts -P inject nit.xml=6000 --replace --pid 16 -P tables -p 16 -a -O drop | grep Section:` 

I have used your xml example for testing.  input.ts file contains PCRs and the huge NIT table which is replaced by the inject plugin.

lelegard, 2020-01-26:
OK, with infinite loop on the input TS and a repetition rate on the section file, I can reproduce the problem.

In fact, during some time, a few input file cycles, the section ordering is ascending as expected. Then, after some point, the section order changes, descending, then ascending, descending again, etc. with some sequences or random order.

All in all, all sections are sent in less than two cycles of the table, so a conformant receiver should be able to get the NIT anyway. But this is certainly not clean. There is probably something wrong in the section scheduling. This may be a bit tricky to investigate. I will have a look.

bobben1no, 2020-01-27:
This is not a big problem. I also think decoders will work fine even with reversed section order.  

It seems like I found a way of controlling the repetition rate with inject bitrate setting instead of inject replace. So far it seems to give normal section order:

`tsp --bitrate 600000 -I file --infinite CBR_TS_with_SI_noPCRs.ts -P filter --negate --pid 0x0010 -P inject --bitrate 12000 --pid 0x0010 --stuffing NIT.xml -P regulate --bitrate 600000 -O ip --rtp --local-address 10.10.10.1 --local-port 5000 238.0.0.55:5000`

The input TS file does not contain any PCRs (maybe that is working better with infinite looping?).  It  was recorded as CBR 600kbps and contains PID 16,17,18 and 8191(stuffing).

commit 0938b298 (2020-04-04): Fixed #515 (wrong order of injected sections)
closed 2020-04-04 by lelegard