tsduck/tsduck #519: Error: dvb: tuning error on /dev/dvb/adapter5/frontend0: Invalid argument

j3dd4h, 2020-02-01:
I have TBS6909 DVB-S PCIe tuner with dvblast ( dvblast -f 3786000 -s 5632000 -v 18 -a 4) is working but failing when using tsduck.

_root@dvb-s# **tsp --debug -I dvb -a 5 --lnb 5150 --frequency 3786000000 --symbol-rate 5632000000  --polarity vertical --delivery-system DVB-S --modulation QPSK --satellite-number 0 --spectral-inversion auto  -P until --seconds 10 -O drop**  
tsp: Debug: debug level set to 1
* Debug: debug level set to 1
* Debug: trying to load "/usr/local/bin/tsplugin_dvb.so"
* Debug: symbol tspNewOutput not found in /usr/local/bin/tsplugin_dvb.so
* Debug: symbol tspNewProcessor not found in /usr/local/bin/tsplugin_dvb.so
* Debug: trying to load "/usr/local/bin/tsplugin_drop.so"
* Debug: symbol tspNewInput not found in /usr/local/bin/tsplugin_drop.so
* Debug: symbol tspNewProcessor not found in /usr/local/bin/tsplugin_drop.so
* Debug: trying to load "/usr/local/bin/tsplugin_until.so"
* Debug: symbol tspNewInput not found in /usr/local/bin/tsplugin_until.so
* Debug: symbol tspNewOutput not found in /usr/local/bin/tsplugin_until.so
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dvb: using /dev/dvb/adapter5 (DVB-S, DVB-S2, DSS)
* Debug: dvb: starting discarding frontend events
* Debug: dvb: finished discarding frontend events
* Debug: dvb: tuning on /dev/dvb/adapter5/frontend0
* Debug: dvb: 1 DTVProperties:
* Debug: dvb: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
* Debug: dvb: starting discarding frontend events
* Debug: dvb: finished discarding frontend events
* Debug: dvb: tuning on /dev/dvb/adapter5/frontend0
* Debug: dvb: 7 DTVProperties:
* Debug: dvb: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 5 (0x00000005)
* Debug: dvb: [1] cmd = 3 (DTV_FREQUENCY), data = 1364000 (0x0014D020)
* Debug: dvb: [2] cmd = 4 (DTV_MODULATION), data = 0 (0x00000000)
* Debug: dvb: [3] cmd = 8 (DTV_SYMBOL_RATE), data = 1337032704 (0x4FB18000)
* Debug: dvb: [4] cmd = 9 (DTV_INNER_FEC), data = 9 (0x00000009)
* Debug: dvb: [5] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
* Debug: dvb: [6] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
* Error: dvb: tuning error on /dev/dvb/adapter5/frontend0: Invalid argument
* Debug: Report logging thread terminated_

root@dvb-s# tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.20-164
root@dvb-s# uname -a
Linux dvb-s 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
root@dvb-s# 
root@dvb-s# tslsdvb 
/dev/dvb/adapter0 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter1 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter2 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter3 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter4 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter5 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter6 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
/dev/dvb/adapter7 (TurboSight TBS 6909 DVB-S/S2 , DVB-S, DVB-S2, DSS)
root@dvb-s#

lelegard, 2020-02-03:
The following parameter is incorrect: `--symbol-rate 5632000000`

5 giga symbols per seconds is impossible on DVB-S. Please pay attention to units. Your command `dvblast -f 3786000 -s 5632000` probably uses kHz for frequency and symbols/s for symbol rate. TSDuck uses "natural" units, i.e. Hz and symbols/s.

You use `--lnb 5150 --frequency 3786000000`. Are you sure about your LNB ?

j3dd4h, 2020-02-04:
Thanks! It's working now.
Btw we using C-BAND so lnb is 5150.

closed 2020-02-04 by j3dd4h