tsduck/tsduck #522: Compilation error when creating a PORTABLE version

lars18th, 2020-02-01:
Hi @lelegard ,

This is related to #511, but is different.

When using different TSDuck versions, I create a subdirectory in the main directory (`~/tsduck-ver-X.XX/bin`). And I copy all binaries into this directory. Then I can use the tools using the full path. And then I don't mix versions as every binary directory is inside a different tree.

However, when a _new_ subdirectory exists the compilation fails. I prepared some fix with the PR #521 . However, this doesn't fix all the problems. So, please can you check the building in Linux when the user creates some subdirectory? I find it very annoying that this simple task completely breaks up the compilation.

Thank you!

lars18th, 2020-02-01:
And another question (I prefer to not open a lot of issues with related things):

- Can you add support for using some custom LD_LIBRARY_PATH, or use the current one?

When using the files `./src/utest/release-x86_64/setenv.sh` and `./src/tstools/release-x86_64/setenv.sh` I need to manually edit them when I use the GCC-4.9 compiler. The reason is because the libraries are installed in a non-standard path (to not mix different compilers, i.e. `/usr/lib/gcc-4.9-backport/lib`).

So the idea is to support something like: `make CUSTOM_LD_LIBRARY_PATH=<custom-dir>`; or to add to these files something like:
```
export LD_LIBRARY_PATH="/<install-dir>/src/libtsduck/release-x86_64:$LD_LIBRARY_PATH"
```

What you think?

lelegard, 2020-02-01:
The `setenv.sh` file is just a trick to run applications directly from the build. This is typically a debug trick (make, run, change code, make, run, etc.) This trick is needed just because the library, the plugins, the applications and the utests have different source locations and different object locations.

If you have several coexisting versions, it is simpler to just copy all binary files in the same directory. If everything is in the same directory, there is no need for `setenv.sh`.

The Unix make-based build is too complicated, I think that instead of having more setenv, more environment variables, I should simplify by compiling into one single binary directory, outside the source tree.

And BTW, to copy everything in the same directory, the command is `make install SYSPREFIX=...`

lars18th, 2020-02-02:
> I should simplify by compiling into one single binary directory, outside the source tree.

Great!

> And BTW, to copy everything in the same directory, the command is `make install SYSPREFIX=...`

I know it, but in my environment this generates other side effects. So for this reason I use my personal script. Perhaps a new update will remove the problems.

Futhermore, one comment: The idea to use `export LD_LIBRARY_PATH="/<install-dir>/src/libtsduck/release-x86_64:$LD_LIBRARY_PATH"` in the two `setenv.sh` files is not to simplify my work. It's only to enable the execution of the tests. So, if you improve the current compilation scripts, and compile all into one directory, then I don't need any modification. Only remember to use the environment LD_LIBRARY_PATH when you compile the library _and_ the tests.

Regards.

closed 2020-03-04 by lelegard