tsduck/tsduck #527: rookie question: how to enable SRT

J-Rogmann, 2020-02-11:
Ahoi
I got a little rookie question on how to enable SRT when building tsduck from source. 
I'm on Ubuntu 18.04 and did **NOT** specify `NOSRT=1` as build option. But still it says `* Error: srt: This version of TSDuck was compiled without SRT support` when trying to use SRT protocol.
My steps were:
```
git clone https://github.com/tsduck/tsduck.git
build/install-prerequisites.sh
make NOPCSC=1 NOCURL=1 NODTAPI=1
source src/tstools/release-x86_64/setenv.sh
```
Are there any special instructions to enable SRT support prior to building? I noticed, for windows there is a special script to download a SRT library build called `install-libsrt.ps1 ` . Is there something equivalent to do on Linux?

best regards,
Justus

lelegard, 2020-02-11:
Hi @J-Rogmann,

Ubuntu starts to support SRT in its official repo with 19.04 (package `libsrt-dev`, version 1.4.1).

With Ubuntu 18.04, there is no SRT base support. If you cannot upgrade Ubuntu, then you have to rebuild and install libsrt yourself (see https://github.com/Haivision/srt).

Once you have manually installed libsrt, just restart `make`. Libsrt will be automatically discovered and used in TSDuck. Note: start with `make distclean` to ensure you restart the build from scratch.

lelegard, 2020-02-11:
>  I noticed, for windows there is a special script to download a SRT library build called install-libsrt.ps1 . Is there something equivalent to do on Linux?

On Linux and macOS, the script `install-prerequisites.sh` installs the appropriate libsrt package when there is one for the system on which it is executed (none for Ubuntu 18.04).

lelegard, 2020-02-11:
Last advice: use `make -j10` or any appropriate number of parallel jobs. A pure sequential `make` is awfully long........

lelegard, 2020-02-11:
Oops, @J-Rogmann, I did not realize you were part of the Haivision/srt team. I assume there is no need to tell you how to build libsrt 😉

J-Rogmann, 2020-02-11:
Well, as I said: I'm still a rookie when it's comes to software DEV :)
I simply forgot to perform a `make install` on the SRT lib. 
Still worth noting for anybody else walking down this route: I still had to set the correct path to the SRT library for TSduck:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/
```
But still I have issues:
```
tsp -I file --infinite /home/elgono/Videos/StressTest_4KP25_h264.ts -P regulate --bitrate 55933496 -O srt 192.168.2.49:4900
* Error: srt: error during srt_bind(), msg: Connection setup failure: unable to create/configure SRT socket: Cannot assign requested address
```
But that I'll discuss with my colleagues. Looks like SRT issue to me.

and thanks for your tip with `make -j10` that feels like jumping from my 3 Cylinder Skoda into my V12 BMW :)

lelegard, 2020-02-11:
> Still worth noting for anybody else walking down this route: I still had to set the correct path to the SRT library for TSduck:
>
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/`

Official packages which come with distros always install in `/usr/lib` and there is nothing to setup. If you manually install _any_ library in `/usr/local/lib`, then you need to setup `LD_LIBRARY_PATH`, regardless of TSDuck or SRT.

J-Rogmann, 2020-02-11:
It works now! Thanks @lelegard for your help! I went back to last official release of SRT v1.4.1 and that solved the problem with the socket.

I would like to share some additional notes here regarding usage of SRT output:

There are some differences with the TSduck SRT implementation compared to other SRT applications. Typically you only specify a port without IP like `srt://:4900` and a SRT-listener on port 4900 will be created. If you specify an IP:port combination, SRT typically would act as SRT-caller to that IP:port (e.g. `srt://192.168.2.49:4900`)

However, with TSduck it seems you only can create a listener port and need to specify the adapter, which it should listen on. In this test the linux machine has two interfaces and the one with IP `192.168.2.3` was chosen to act as listening interface.

```
tsp -I file --infinite /home/n00b/Videos/StressTest_4KP25_h264.ts -P regulate --bitrate 55933496 -O srt "192.168.2.3:4900" --transtype live --messageapi
```

A detail worth mentioning: `-O srt "IP:Port"` for `tsp` already specifies the usage if the srt protocol. So there is no need to add the `srt://` prefix to the IP or DNS-name. Adding it would actually result in an error.

Now another SRT application can connect to that SRT stream on `192.168.2.3:4900` provided by `tsp` (for example a srt-live-transmit demo application to flip it back to a MPEG-TS).

```
srt-live-transmit.exe srt://192.168.2.3:4900 udp://192.168.2.49:10002
```

Or VLC player: press `CTRL+N` to open a new network stream and type in URL `srt://192.168.2.3:4900` to connect to the stream in this example.

TSduck is a great tool and I can only recommend it. Big helper for quick testing SRT connections by playing back .ts files as a stream, if you need reproducible results or have no encoder handy.

best regards,
Justus

lelegard, 2020-02-11:
Thanks for your feedback. This should be discussed with @spnngl. He is the author of the SRT support for TSDuck. I am pretty ignorant in the various possibilities of SRT.

spnngl, 2020-02-12:
Hello, to simplify the TSDuck implementation I made the SRTInputPlugin caller only and the SRTOutputPlugin listener only (except for rendezvous mode supported by both).

I'm not an expert of SRT either, I just needed this protocol and liked a lot TSDuck capabilities so I contributed ^^

To listen on your localhost you can specify port only with TSDuck like this:
`tsp -I ... -P ... -O srt 4242`

Same thing if you call localhost for input:
`tsp -I srt 4242 -P ... -O drop`

Feel free to ask if you have question and modify the TSDuck implementation if I missed something.

cdunklau, 2020-02-19:
> Ubuntu starts to support SRT in its official repo with 19.04 (package `libsrt-dev`, version 1.4.1).

Hi @lelegard, just a small correction, it looks like Ubuntu's libsrt1/libsrt-dev packages are [somewhat older than that](https://packages.ubuntu.com/search?keywords=libsrt) -- 19.04 ships SRT 1.3.1, and 19.10 ships 1.3.2 (unless I've misunderstood something).

Would you mind updating your comment so others don't get confused like me?

lelegard, 2020-02-23:
> just a small correction, it looks like Ubuntu's libsrt1/libsrt-dev packages are somewhat older than that -- 19.04 ships SRT 1.3.1, and 19.10 ships 1.3.2 (unless I've misunderstood something).

Sorry, my fault, you are perfectly right. SRT 1.4.1 is shipped with Fedora 31. Ubuntu usually lags behind Fedora for almost everything.

> Would you mind updating your comment so others don't get confused like me?

Done.

jbieberly, 2020-05-28:
I have SRT working fine, but there is some functionality I believe is missing and I haven't been able to get any of the options to perform a restart on loss of connection. 

Here is the command I'm using:

`tsp` --ignore-joint-termination -v -I file file.ts --infinite -P regulate -O srt 12345 --transtype live `--messageapi`

The current behavior is, I run this command, then connect to the machine with VLC, which plays the video successfully, then if I close VLC, the command has stopped / timed out.  

I would like the TSP command to continue running regardless of connection status, and be able to accommodate a re-connection after a failure.  Is this possible?

spnngl, 2020-05-28:
> I would like the TSP command to continue running regardless of connection status, and be able to accommodate a re-connection after a failure.  Is this possible?

Hi, when I have done the SRT plugin for TSDuck my use case did not need reconnection. So I made a simpler implementation: it connects, transfers and disconnects when it is done/lost

If you want it you can add it to TSDuck, it is always good to add functionality :)

D3nis, 2020-10-30:
Nice time.

I builded tsduck with libsrt , but get this erros: 

```
#tsp --realtime=yes --verbose -m --timed-log -P bitrate_monitor -p 60 -I srt --rendezvous feed-src:22265 --transtype live -P continuity -O ip  231.13.22.65:1234
* 2020/10/30 20:53:56 - Error: srt: This version of TSDuck was compiled without SRT support
* 2020/10/30 20:53:57 - Error: srt: missing parameter

```

TSduck  version 3.24-2042

# rpm -qa | grep libsrt
libsrtp-1.5.4-8.el8.x86_64
libsrtp-devel-1.5.4-8.el8.x86_64

Your  script  build/install-prerequisites.sh was also executed.

make NODTAPI=1 NOPCSC=1 NOTELETEXT=1

# uname -a
Linux feed-am  4.18.0-193.14.2.el8_2.x86_64 #1 SMP Sun Jul 26 03:54:29 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# cat /etc/redhat-release 
CentOS Linux release 8.2.2004 (Core)

# grep -i srt  make_output_tsduck.txt
  [DEP] plugin/tsSRTInputPlugin.cpp
  [DEP] plugin/tsSRTOutputPlugin.cpp
  [DEP] base/network/tsSRTSocket.cpp
  [CXX] plugin/tsSRTInputPlugin.cpp
  [CXX] plugin/tsSRTOutputPlugin.cpp
  [CXX] base/network/tsSRTSocket.cpp

How I can troubleshoot this?

lelegard, 2020-10-30:
You have SRTP installed (`libsrtp-devel`), not SRT (`srt-devel`).

SRT is relatively recent and not available on all distros.

To make sure you have all required dependencies before building, you can execute the script `build/install-prerequisites.sh`. It installs whatever is necessary to build TSDuck and is available on the current system. I try to maintain it but it is sometimes hard to be aware in a timely fashion of what's new on which version of which distro that could be useful to build TSDuck. The last time I checked, SRT was not available on CentOS / RHEL (only on Fedora and Ubuntu). But trying on my CentOS 8.2 VM, I just realised it was available now. I will update `install-prerequisites.sh` accordingly.

In the meantime, just run `dnf install srt-devel`, cleanup TSDuck (`make distclean`) and rebuild (`make -j10`).

D3nis, 2020-10-31:
Thank You Lelegard,

I just searched for "srt" (dnf search srt) but didn't take a look to the package self.

please add to Your install-prerequisites.sh for Centos8 this line:
`dnf config-manager --set-enabled PowerTools`

because of this packages: doxygen pcsc-lite . Else they can not be found without enabled PowerTools.

Thank You for Your time and nice project!

lelegard, 2020-10-31:
Thanks, PowerTools was already enabled on my CentOS VM's and I did not notice the explicit dependency.
The script has been fixed.