tsduck/tsduck #528: Issue Decrypting AES-CBC Encrypted Multicast Stream.

seanpdiaz, 2020-02-12:
I am experiencing an issue when trying to decrypt an AES-CBC encrypted multicast stream. The below command is what I am using when trying to decrypt the stream.

`tsp --verbose -P aes --cbc --descramble --key 33333333333333333333333333333333 \
-p 101 \
-I ip 239.192.0.80:5000 --buffer-size 50000000 -O ip 239.193.1.80:6000`

The result that I experience when running this command is that I am able to open the stream in VLC however it displays nothing but macroblocking ([Seen Here](imgur.com/a/kFyilUZ)). I am not quite sure what I am doing wrong here but any help would be much appreciated. I can provide a pcap of the encrypted stream if that is helpful.

lelegard, 2020-02-12:
Hi @seanpdiaz,

You did not _exactly_ describe how was the stream ciphered. Saying AES-CBC is not sufficient to correctly decipher the stream. You also need to know the initialization vector and the residue processing. One or both must be incorrect. This is probably the reason for your problem. 

The `aes` plugin is described in the documentation as _experimental_. It is used to experiment the various possible ways of doing AES scrambling on transport streams and all crypto options must be carefully selected. With option `--aes-cbc` alone, the initialization vector is zero and the residue is left clear by default. This is were your encryption probably differs. 

In the real world, experimentation left aside, the two standards of AES-CBC encryption on transport streams are ATIS-IDSA and DVB-CISSA. The former is used in IPTV, its initialization vector is zero and its residue processing is similar to DVS042. The latter uses a hard-coded non-zero initialization vector and the residue is left clear. 

If your stream uses one of these two standards (more likely ATIS-IDSA if this is a multicast stream), you’d rather use the `descrambler` plugin instead of the `aes` one. The option `--atis-idsa` or `--dvb-cissa` will directly provide the right descrambling configuration.

seanpdiaz, 2020-02-13:
Thanks this issue has been resolved and was in fact a misconfiguration on my part. Thanks for the help and for the great software.

closed 2020-02-13 by seanpdiaz