tsduck/tsduck #532: FILE input: Negative offsets?

lars18th, 2020-02-17:
Hi @lelegard ,

I want to _join_ (mux with the interleave mode) two files, but I want to insert some delay between them. Using the `FILE` input different offset parameters are implemented: `--byte-offset` and `--packet-offset`. However the values are POSITIVE from 0 start.

However, if we want to start with one file at 0, and after some time (N bytes) start the other, then it's necessary that the offsets can be negative. In this case, the implementation could be simple: read NULL packets while the offset is negative.

Do you think that has sense?
Regards.

lars18th, 2020-02-17:
Hi @lelegard ,

The request is more or less, like the parameter `--add-start-stuffing` for `tsp`, but for the `FILE` input for each filename.

I hope you want to consider this.

Regards.

lars18th, 2020-02-17:
Hi @lelegard ,

The final suggestion is to add to the `FILE` input the parameters `--add-start-stuffing` and `--add-stop-stuffing` for each filename. If you can provide it then it will be very useful.

Thank you for accepting a lot of our requests!

lelegard, 2020-02-23:
In the general case, there are N input files, not only 2. There is no reason to use the same number of null packets before / after each file. So, we would need to invent a complicated syntax to specify "n1 packets before file 1", "n2 packets before file 2", etc.

Since the input are files, it is a lot easier to pre-process files which need initial stuffing with something like:
~~~
tsp --add-start-stuffing 100 -I file file2.ts -O file file2.stuff.ts
tsp -I file file1.ts file2.stuff.ts ...
~~~
You can also automate this in a shell script. Remember that TSDuck is essentially a scripting tool.

lars18th, 2020-02-23:
Hi @lelegard ,

First: Good to see you after a few days' rest! I hope you're okay.

> Since the input are files, it is a lot easier to pre-process files which need initial stuffing with something like:

Wel, yes and no. Some of my files are FIFOs and others are local HLS chunks. So is not the best to _preprocess_ all the time. Futhermore because I can't pass more than one stream using PIPEs I'm searching for other solutions.

So, my request can be something like: `-I file --add-start-stuffing 0,1000,256 file1.ts file2.ts file3.ts [...]`. I feel is not complicated and it's easy to parse.

Perhaps the use case will be help you to understand the reason of this request: I'm using the FFmpeg to encode more than one program, and I'm using the TSDuck to mux the different streams inside one MPTS. And in this particular case it's right to not be starting all GOPs at the same time. So I need to mark different start points using _random_ NULL stuff at the start of each program. That's the key point.

So, please consider this "simple" `--add-start-stuffing [value_file1][value_file2],[value_file3],[...]` syntax for the FILE input.

Thank you.

lars18th, 2020-11-25:
Hi @lelegard ,

Please, can you consider my last request of the "simple" enhancement syntax?
` --add-start-stuffing [value_file1][value_file2],[value_file3],[...]`

(The default value is 0, so if the number of input files is greater than the stuff values, then no ambiguity is created).

Perhaps you think this is irrelevant, but for (hard) muxing using files that can't be preprocessed this will be very useful.
(Note: The concept of "hard muxing" is the _combination_ of multiple input files using the `file` input).

I hope you want to consider this enhancement.
Thank you!

lelegard, 2020-11-25:
The option `--add-start-stuffing` applies to `tsp`. And `tsp` has no idea of "input files". It just activates an input plugin. The "start stuffing" packets are inserted before reading the first packet of the input plugin, whatever it is.

lelegard, 2020-11-27:
Added options `--add-start-stuffing` and `--add-stop-stuffing` to plugin `file`.

lars18th, 2020-11-27:
> Added options `--add-start-stuffing` and `--add-stop-stuffing` to plugin `file`.

Thank you @lelegard !
I see that instead of delimiters, you implemented support of multiple ocurrences. So, my example will be finally:
`-I file --add-start-stuffing 0 --add-start-stuffing 1000 --add-start-stuffing 256 file1.ts file2.ts file3.ts [...]`
Great! 👍 

It's a pleasure that you're listening to all our requests. I'm trying to ask only for improvements that make sense. And I see that you're a very reasonable person.
Thank you!

closed 2020-11-27 by lars18th

lelegard, 2020-11-27:
> I see that instead of delimiters, you implemented support of multiple ocurrences. 

First, multiple occurrences is more consistent with the rest of TSDuck. Second, commas as interpreted as thousands separators in integer values and are ignored. So 1,2,3 would be interpreted as 123.