tsduck/tsduck #534: Trouble: PSI-MERGE multiple times

lars18th, 2020-02-17:
Hi @lelegard ,

I have a trouble with the `psi-merge` plugin after a `FILE` input with `--interleave`.
Here the example:
```
tsp -I file --interleave=1 --label-base 2 --first-terminate file1.ts file2.ts file3.ts file4.ts \
 -P psimerge --main-label 2 --merge-label 3 \
 -P psimerge --main-label 2 --merge-label 4 \
 -P psimerge --main-label 2 --merge-label 5 \
 -O file output.ts
```

All input files are equal: CBR generated with FFmpeg, and without overlaped pids/services. So only PAT, SDT, PMT and A/V pids are present.

Then the expected behaviour is to merge all PSI/SI tables. Correct? Not at all!
- For the PAT table, one initial ocurrence with version 0 appears in the output with only the services of `file1.ts`. After that a correct table with version 2 is found (with all services from the four input files).
- For the SDT, two incorrect occurrence appears. First one with version 0 and services only from `file1.ts`. After another occurrence with version 1 and services from `file1.ts` and `file4.ts`. And after that the correct table with version 2 and all services.

Is this the expected behaviour? How I can remove these "ancient" tables?
Please help me.
Thank you!

lars18th, 2020-02-17:
Hi @lelegard ,

Regarding this trouble, my first trial to overcome this problem is to remove tables with the old version. However, I don't know how to select packets of an specific version. It exists one filter to target this functionality?

Regards.

lars18th, 2020-02-18:
Hi @lelegard ,

> Is this the expected behaviour? 

After thinking on your comments about other functionalities in the TSDuck, I feel this is the expected behaviour. That's because it works focused on realtime and after the initial tables, the values are correct. So don't try to fix this.

> How I can remove these "ancient" tables?

However, I feel will be good to have a method to _remove tables with specific version numbers_. So I suggest to add a new plugin called `table-filter`. The functionality will be in pair with the `filter` plugin but focused on _tables_ and not in _packets_. Examples:

- `table-filter --pmt --version 0-2`: This filter out all packets of a PMT table with versions 0, 1 or 2.
- `table-filter -n --pat --version 2 --version 5`: This pass only packets of PAT tables with version 2 or 5.
- `table-filter --any --version 3`: Filter out all tables with version 3.
- `table-filter --eit --table_id 0x50`: Remove tables with schedule info from EIT.

What you think about this proposal?

lars18th, 2020-03-12:
Hi @lelegard ,

Until this enhancement can be partially implemented, any idea about how to do `table-filter --eit --table_id 0x50` ?
I want to remove the schedule EIT and leave the p/f EIT. It's possible to do it at time?

Thank you!

lelegard, 2020-03-12:
What about using plugin `sections` ?

lars18th, 2020-03-17:
> What about using plugin `sections` ?

Thank you! :+1: 

Then to filter out (aka remove) the schedule EIT of the service 0x6f81, this will work?
`tsp ... -P sections -e  0x506f81 -p 18`

As the documentation the parameter `-t 0x50-0x59` removes all schedule info, but for all services. The parameter `-e` can be used to filter out only for one service?

Regards.

lelegard, 2020-03-18:
> Then to filter out (aka remove) the schedule EIT of the service 0x6f81, this will work?
> `tsp ... -P sections -e 0x506f81 -p 18`

Yes, except that you have to enumerate all variants with table ids from 0x50 to 0x5F (16 values): `-e 0x506f81 -e 0x516f81 ... -e 0x5f6f81`

>As the documentation the parameter -t 0x50-0x59 removes all schedule info, but for all services. The parameter -e can be used to filter out only for one service?

Yes, each `-e` option applies to one pair of table id / table id extension, here a pair of EITs / service id. You must enumerate all services to remove, for all possible EIT schedule id. As usual with TSDuck, you'd better use a shell script, for instance:
```
opt=""
for srv in 0xDEAD 0xBEEF 0x0F00; do
    for ((tid=0x50; $tid<=0x5F; tid++)); do
        opt="$opt$(printf ' -e 0x%X' $(( ($(($tid)) << 16) | $(($srv)) )))"
    done
done

tsp ... -P sections $opt ...
```

lars18th, 2020-11-25:
Hi @lelegard ,

Only for _convenience_ I suggest a small change in the parameters of the filter `sections` for overcome the use of a _very large_ commands (I know you don't want to reinvent the wheel, but...):
```
Usage
tsp -P sections [options]

Options
  -e id1[-id2]
  --etid-remove id1[-id2]
    Remove all sections with the corresponding extended table id values. The value is a combination
    of the table id and the table id extension.
    For example, the option -e 0x4A1234 removes all BAT sections (table id 0x4A) for bouquet id
    0x1234 (table id extension).
    Several options --etid-remove can be specified.

  --etid-remove-reverse id1[-id2]
    Remove all sections with the corresponding extended table id values in reversed order. The value
    is a combination of the table id extension and the table id.
    For example, the option -e 0x12344A removes all BAT sections (table id 0x4A) for bouquet id
    0x1234 (table id extension). And -e 0x123450-0x12345F removes all scheduled EIT (0x50-0x5F) for
    service id 0x1234.
    Several options --etid-remove-reverse can be specified.
```

I hope you agree with this _small_ enhancement.
Regards.

lelegard, 2020-11-25:
Added option `--reverse-etid`

[image]

lars18th, 2020-11-25:
> Added option `--reverse-etid`

Thank you, @lelegard ! :+1: 
I appreciate very highly that you listen to the suggestions we make.

So finally, and before closing this this issue, any idea to do what I commented time ago?

> I feel will be good to have a method to _remove tables with specific version numbers_.

lelegard, 2020-11-26:
Added option `--version-remove`

lars18th, 2020-11-27:
Hi @lelegard ,

> Added option `--version-remove`

I see that you've implemented a lot more, including "and" operators. So I need to check it before close this issue.

Thank you for all your effort!

closed 2021-06-03 by lars18th