tsduck/tsduck #535: decoding incorrectly

phamduythai92mta, 2020-02-18:
Hi @lelegard ,
i use this command line to decap a tsfile:
tsp -I file (my_file) -P  decap -p 0x0C1C  -P filter --pattern (my_pattern)  --search-payload 
and a lot of places the text was decoded like that.
in addition i wanna save the output to a txt file, 
tsp -I file (my_file) -P  decap -p 0x0C1C  -P filter --pattern (my_pattern)  --search-payload > myfile.txt
however i can not read it with my python program, it is not utf-8 format

some output is like the following lines ������������������������������������������������������������������������������������������������������������������������������������G␊├=GBK
C┼├␊┼├-L▒┼±┤▒±␊: ≥␤-
C┼├␊┼├-L␊┼±├␤: 197
D▒├␊: F␋, 06 D␊␌ 2019 06:42:02 GMT

what i was wrong?

many thanks in advance!

lelegard, 2020-02-23:
First, what kind of encapsulation are you trying to decapsulate? The `decap` plugin uses some kind of proprietary encapsulation which is implemented by `encap`. The two plugins are working together only. If your stream is encapsulated using some standard protocol, you must use the corresponding decapsulation mechanism.

Second, transport streams are binary. Storing them is a `.txt` will not make them UTF-8 content. What did you expect? The command `tsdump` dumps transport stream packets in text format, but this may not be what you expect.

phamduythai92mta, 2020-02-25:
thanks for your reply!
when i open it by wireshark, as you can see that is MPE encapsulation. now i wanna extract the data on HTTP layer.
when i see in the wireshark, i notice that MPE section divided into 2 TS cells (2819, 2820), that is amazing that wireshark can reassemble MPE section and dissection the data. 
with TS duck i can extract the data on HTTP layer like Wireshark does?
in addition i wanna create a program to do that, can i use TSduck as a library? or any suggestions?
[image]

lelegard, 2020-02-25:
OK, you did not mention MPE. As I mentioned, there are many different kinds of data encapsulations, with many different formats. And there are many different plugins for many different usages. Please read the user's guide. 

To extract MPE-encapsulated data, the plugin is named `mpe`. The encapsulated data are UDP datagrams, not HTTP (which is a TCP-based protocol). Several distinct streams can be encapsulated in the same PID, you may select which ones to extract. The extracted datagrams can be either saved in a file or reforwarded on UDP. Remember that those UDP datagrams may contain TS packet or anything else.

TSDuck is a low-level toolbox which requires some knowledge of the various layers of protocols which are manipulated.

phamduythai92mta, 2020-02-25:
sorry about my insufficient knowledge!
Now i can extract the UDP payload with MPE. (but the UDP payload is not what i need)
However, i am just curious about that why only UDP payload? how about TCP payload? can TSDuck do extract TCP payload?
you can see the picture i posted, my ts file (MPE->IP ->TCP->HTTP), this is actually MPE encapsulation, but the encapsulated data are TCP datagrams. so could you kindly suggest any solution to extract the TCP payload.
many thanks!

lelegard, 2020-02-25:
Hi,

MPE is designed to carry IP datagrams. Potentially, it can carry any type of IP datagram. But, in practice, you only have UDP datagrams.

You cannot have TCP inside MPE, this is a nonsense. MPE is encapsulated into a transport stream. TS is a one-way, non-reliable and one-to-many transport. TCP is a duplex, reliable, one-to-one protocol. Even if data go one way only at application level, there are constant exchanges at TCP level. Typically, the sender would stop sending after filling the "TCP window size" without acknowledgement from the peer. With a TS, there is no response. There is not even one single peer.

So, _by design_, TCP cannot be sent through MPE. And since HTTP is a TCP-based protocol, it cannot be sent either.

phamduythai92mta, 2020-02-25:
sorry if i have lots of questions and thanks for broadening my knowledge!
it is strange that my TS file is a real file captured in practice and now i have to work on it.
any suggestion to solve my problem?

lelegard, 2020-02-25:
> any suggestion to solve my problem?

You cannot solve a problem before having clearly defined it.
First, you have to define your problem:
- What is the "file"? Which format?
- Which application inserts in the MPE in the TS?
- What kind of IP datagrams are inserted? Where do they come from?
- What are the layers of protocols?
- Which kind of data would you like to extract from the MPE stream?
- What do you want to do with the data?

Once you have answered these questions, you can start looking for a technical solution. I cannot help you answering those questions. This is your file, not mine. I have no idea on what is inside it.