tsduck/tsduck #536: BUG: --add-start-stuffing and PIPES

lars18th, 2020-02-18:
Hi @lelegard ,

I found a bug (or almost a problem) using TSDuck and FFmpeg. Two similar commands:
- `ffmpeg [...] pipe:1 | tsp --add-start-stuffing 40000 [...] -O file test.ts`: OK
- `ffmpeg [...] pipe:1 | tsp --add-start-stuffing 50000 [...] -O file test.ts`: FAIL

The error appears in ffmpeg as `Error writing trailer of pipe:1: Invalid argument` and I feel is related to the impossibility to write to the PIPE.

Please, can you check why for values over 40000 packets (I don't know the exact number, but the boundary is between 40-50k) of add-start-stuffing the PIPE fails?

Thank you!

lars18th, 2020-03-05:
Hi @lelegard ,

Any idea about this issue?

lelegard, 2020-03-06:
Fixed in commit 9a04802c14e1f6485511e4df3ab407d662197417.

This is a bug. When `--add-start-stuffing` was so large that it completely filled the initial buffer (half the global buffer size), then an end of file was incorrectly assumed because nothing could be read from the input plugin since only stuffing was generated.

The actual limit was 44619 packets, 8 MB, half the default size of the global buffer. The following command worked. But it failed with 44620. The failure was silent because `tsp` considered that it had reached the end of input, which is not an error.
~~~
tsp --add-start-stuffing 44619 -I null 1 -O file tmp.ts
~~~

Note that you could make it work by increasing `--buffer-size-mb` to more than twice the `--add-start-stuffing`.

The problem has nothing to do with `ffmpeg`. Since `tsp` terminated, the pipe was closed and `ffmpeg` could not write into it.

The original idea is that if the input is empty or error, no output should be generated at all. So, the additional start stuffing is dropped. With the new fix, this is still the case but only when the initial start stuffing is less that half the global buffer. If you specify more than half the global buffer, then the output will contain most of the initial stuffing even if the input is empty or error.

lars18th, 2020-03-06:
Hi @lelegard ,

Thank you! So we need to assume that the workaround is to check the `--buffer-size-mb` every time large values are used for `--add-start-stuffing`. It's right this assumption? Then I suggest a small change in the code: when the `--add-start-stuffing` will be greater than a half of the input buffer less one, then print a warning when running the `tsp` tool. Something like _"WARNING: Input buffer with insufficient size for the --add-start-stuffing parameter."_.

Regards.

lelegard, 2020-03-06:
The workaround is for the buggy version. The code has been fixed in the repo to allow large `--add-start-input`. So, there won't be any problem.

lelegard, 2020-03-16:
Fixed

closed 2020-03-16 by lelegard