tsduck/tsduck #537: BUG? TABLES --xml-output empty for EIT

lars18th, 2020-02-19:
Hi,

I don't know why this generates an **empty** file:
-`tsp -v -I ip 239.0.0.1:12345 -P tables -p 18 --all-once -f --xml-output test.xml -O drop`
And this works:
-`tsp -v -I ip 239.0.0.1:12345 -P tables -p 18 --all-once -f --text-output test.txt -O drop`

As I can see in the Documentation the EIT Class has the `toXML()` member:
https://tsduck.io/doxy/classts_1_1_e_i_t.html

Any idea?

lars18th, 2020-02-19:
Hi @lelegard ,

This example generates the same problem:
```
tstabcomp -v -d eit.bin eit.xml
Decompiling eit.bin to eit.xml
tstabcomp: eit.bin: 709 orphan sections not saved in XML document (0 tables saved)
tstabcomp: cannot decompile XML file eit.xml
```

But after reading this post from you: https://github.com/tsduck/tsduck/issues/160#issuecomment-437808276 I discovered the _REQUIREMENT_ to use `--pack-all-sections` when using `tstables` or `-P tables` with EIT.

So, please can you include some WARNING in the documentation and the help?

Regards.

lelegard, 2020-02-23:
This is due to the weird nature of EIT's. EIT's have the potential to be complete and consistent tables. But, when the guidelines from ETSI TS 101 211 are applied, then EIT's are no longer "tables" but sparse sections. You can view sections, but there is no complete EIT table, as defined by ISO 13818-1.

The option `--pack-all-sections` is just a trick to transform sparse sections into tables.

And this is already documented:
~~~
  --pack-all-sections
      Same as --all-sections but also modify each long section so that it
      becomes a valid complete table. Its section_number and
      last_section_number are forced to zero. Use with care because this may
      create inconsistent tables. This option can be useful with tables with
      sparse sections such as EIT's to save them in XML format (as an
      alternative, see also --fill-eit).
~~~
I do not see what to say more...

lars18th, 2020-02-23:
Hi,

Perhaps this simple change can be useful:
```
  --pack-all-sections
      Same as --all-sections but also modify each long section so that it
      becomes a valid complete table. Its section_number and
      last_section_number are forced to zero. Use with care because this may
      create inconsistent tables. This option can be REQUIRED with tables with
      sparse sections such as EIT's to save them in XML format (as an
      alternative, see also --fill-eit).
```
Regards.

closed 2026-04-24 by github-actions[bot]