tsduck/tsduck #538: Question: How to joind from  "-P tables -m"

lars18th, 2020-02-19:
Hi,

I want to do some custom processing of the EIT information. So, I use the `tables` plugin to capture the different tables in a binary form: `tsp [...] -P tables -p 18 --all-once -f -m -b test.bin -O drop`.

Then I can process these files. However, I don't know how to _JOIN_ them! The plugin to insert tables is `inject`. And it can insert multiple files, but when the number of files is large...

Any idea to something like this?
-`tsp [...] -P inject -p 18 --join-section-files test_p0012_*.bin [...]`

Thank you

lars18th, 2020-02-29:
Hi @lelegard ,

Any idea to complete the "join" of the saved binary tables?

Thank you!

lelegard, 2020-02-29:
From the TSDuck user's guide:
> __2.2.1 PSI/SI binary format__
>
> A PSI/SI binary file contains one or more sections in a simple binary format. Each section is directly written in the file without any encapsulation or synchronization information. All sections are contiguous in the file.

So, joining binary section files is simply concatenating them. The magic command you are looking for is consequently:
~~~
cat test_p0012_*.bin  >test.bin 
~~~

lars18th, 2020-02-29:
> So, joining binary section files is simply concatenating them. 

Really? So easy? I'll try.
Thank you!

closed 2020-02-29 by lars18th