tsduck/tsduck #539: make process fails on Ubuntu 18 with GNU Make 4.1

astrokom, 2020-02-19:
Hello @lelegard ,

i tried to build your software with the following command on Ubuntu 18 with GNU Make 4.1:
make NOCURL=1 NOTEST=1 NOPCSC=1 NODTAPI=1 NOTELETEXT=1

The make process fails with the following error message:
make: execvp: /bin/bash: Argument list too long
Makefile:59: recipe for target 'default' failed
make: *** [default] Error 127

The path of my current working directory consists of 67 characters. If i remove one of these characters, the build process is successful.

Also I want to build your software with bitbake and openembedded, but it fails with the same error message due to the long paths.

Kind regards,

astrokom

lelegard, 2020-02-23:
Fixed in commit ae12d05d27267cd310a820c056bdc2e56e3d11c6

In fact, it is a limitation of the Linux kernel in the size of the environment of a created process. The makefile created an environment variable with the list of all source files in the TSDuck library (hundredths of files). With long paths, the total size of the environment was too large.

I have read several discussions arguing that a "modern" POSIX-compliant OS should accept at least 2 MB of environment and command, clearly saying that Linux was not a "modern POSIX-compliant OS" since its limitation is a few kilobytes only. As a matter of fact, I do not see the problem on macOS, a BSD-family Unix system, even though we use bash and GNU make on macOS.

The funny part of the story is that this environment variable was no longer used. So, I cleaned up a couple of makefiles and the problem is gone.

astrokom, 2020-02-25:
Ah perfect, i can confirm that the issue is fixed.
Thank you very much!

closed 2020-02-25 by astrokom