tsduck/tsduck #540: Request: Analyze including PCR intervals

lars18th, 2020-02-24:
Hi @lelegard ,

When using the `tsanalyze` with the wide output, the TABLES sections include: `Repetition`, `Min repet.` and `Max repet.`. However for PIDs with PCR, this data is not provided. But it will be very useful to know the medium and extreme intervals of PCR timestamps. Don't you think?

I hope you want to incorporate it.
Thank you!

lars18th, 2020-02-28:
Hi @lelegard ,

Even if you don't want to touch the `tsanalyze` tool for any reason, perhaps you can consider to improve the `pcrverify` plugin. I need to check the repetition intervals of the PCRs and I don't know how to do it. Any idea?

Thank you!

lars18th, 2020-02-29:
Hi @lelegard ,

Excuse me for asking a lot of questions (and some pretty stupid), but I'm learning a lot about Digital TV using this project.
Any idea to obtain the PCR _repetition_ intervals?

Thank you!

lelegard, 2020-02-29:
The plugin `pcrextract` reports the packet index of each PCR. Knowing the bitrate of the TS, it is easy to compute (or approximate in VBR) the time interval between PCR. The output being a CSV file, it is easy to analyze using a script.

lars18th, 2020-02-29:
> The plugin `pcrextract` reports the packet index of each PCR. Knowing the bitrate of the TS, it is easy to compute (or approximate in VBR) the time interval between PCR. The output being a CSV file, it is easy to analyze using a script.

Great! The inputs are MPTS, so CBR, then the bitrate is fixed and easy to calculate. I feel it will be easy then to compute the intervals. Thank you. :+1: 

However, I suggest (only if you want) to remember to add at some time in the future this capability to this plugin `pcrextract` or to the `tsanalyze` tool.

Thank you @lelegard to provide selfless support.
Regards.