tsduck/tsduck #541: Rouke Question:- How to read the Input TS from Dektec ?

Jazzykool85, 2020-02-25:
I am very new to TSDuck,
I want to monitor Live stream , especially for missing PIDs or descriptors?

As part of this i am trying to read the live TS PSI/SI data including all PIDs and descriptors.
The stream is pushed to the PC using a Dektec DTU -245 demodulator. 
What command i should use to acheive this.
I am looking at the plugin option exactly for Dektec in this case.

lelegard, 2020-02-25:
You have to use the input plugin named `dektec`. The exact options depend on what you are monitoring. Please read the TSDuck user's guide for a list of all possible options for this input plugin.

Then, you have to clearly define what you want to do or analyze from that stream. There is a large number of plugins for various usages. Browse through them to check if any can be useful for what you are doing. Also read the "examples" chapter in the user's guide.

Jazzykool85, 2020-02-25:
Thanks for this response , i will explore more for this plugin options

Jazzykool85, 2020-02-26:
Hi Lelegrand,

Good news, i am able to read the TS Input from dectek and generate a report for PID analysis.
tsp -I dektec -d 0 -P until -s 20 -P analyze --table-analysis --title "R1 HomeTP" -o R1.analysis1 /
 -O drop

But  i am unable to get the individual configured descriptor tags and values for that transponder .
Is there any plugin which can fetch this information.
Many thanks in advance for the guidance 
Thanks
Jazeem

lelegard, 2020-02-26:
> But i am unable to get the individual configured descriptor tags and values for that transponder .

What do you mean by "configured descriptor tags and values for that transponder"?

Jazzykool85, 2020-02-26:
NIT Linkage Descriptors like -> Home TP descriptors, OTA descriptors etc

lelegard, 2020-02-26:
All tables and sections can be displayed using the plugin `tables` or the command `tstables`. There are many options and use case. Please RTFM. To get the NIT use `-P tables --pid 16`.

Jazzykool85, 2020-02-28:
Kool, I am to get it working 
Thanks a lot

closed 2020-06-03 by Jazzykool85