tsduck/tsduck #542: need help

elmofty, 2020-02-25:
hello to all
i have my current udp incoming transport stream and i need to creat new service for network and inject the new service with hex data comming from another udp port i can creat new service but cannot inject incoming udp hex in the service
any one can help me
thanks

lelegard, 2020-02-26:
> new service with hex data comming from another udp port 

I have no idea of what "hex data" means for a service.

elmofty, 2020-02-26:
thanks for your replay
i use this script 
tsp -v --bitrate 11880000 \
 -I ip 13000 \
 -P svremove "Data" --stuffing \
 -P pat --add-service 110/0x0210 \
 -P inject pmt-mpe.xml --pid 0x0210 --bitrate 15000 \
 -P sdt --service-id 110 --name "Data" --provider "data provider" --type 0x0D \
 -P mpeinject 40000 --max-queue 512 --pid 0x0401 \
 -O ip 12345

it can receive the transport stream on udp from port 13000 and receive udb data in port 40000
and inject the udp data in the transport stream and output to udp on port 12345
my problem is i must remove service to inject the new udp data
i want to know haw to creat new pmt with service id without remove service

the udp data i recive like this 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 12 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 12 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 12 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 12 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 12 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 80 81 12 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 91 12 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F 

and it uplink ok but with remove service
i need to inject it without remove service

and this is the pmt-mpe.xml file

<?xml version="1.0" encoding="UTF-8"?>
<tsduck>

<PMT service_id="110">
<component elementary_PID="0x0401" stream_type="0x0D">
<stream_identifier_descriptor component_tag="1"/>
<data_broadcast_id_descriptor data_broadcast_id="0x0005"/>
</component>
</PMT>
</tsduck>

if i remove  -P svremove "Data" --stuffing \    from the script the channel created but the problem in pmt table creat only the SDT but no any ES PID created
thanks for your help

lelegard, 2020-02-26:
To insert data (a full service in your case), you need stuffing because inserting data is done by replacing stuffing. There are several ways to get stuffing in your input stream. The first one is to remove one or more services and replacing them by stuffing, like you do. Another way is to insert additional interleaved stuffing from the beginning using the `tsp` option `--add-input-stuffing`. Read the `tsp` documentation in the TSDuck User's guide for details and examples.

elmofty, 2020-03-01:
thanlks for your support
i have fix the problem and it work well with me
i have another point 
i try to use scrambler like this

tsp -v --bitrate 10880000 \
-I ip 13000 \
-P scrambler "be live 1" \
   --cw 00112233445566778899aabbccddeeff \
   --aes-ctr \
   --iv 0123456789abcdef0123456789abcdef \
   --output-cw-file /home/be_live/cw \
   --log-protocol=info --log-data=debug \
-O ip 127.0.0.1:14000

it working well with me
and i can descrambler it
my problem i need to change the scrambler ce on air without restart
i try but cannot
i must change the cw and restart the tsp
is there any way to do without restart
to change the cw on air
thanks for your support

lelegard, 2020-03-02:
If you need to change the CW on a regular basis (crypto-periods), you can prepare them in a text file and use option `--cw-file`.

Note that `--output-cw-file` is useless here. It is done for the standard DVB SimulCrypt more where CW are randomly generated every crypto-period and inserted in ECM's. If, for test and validation purpose, you need to get the value of the generated CW (other than in encrypted form in ECM's), this option saves all CW in a text file.