tsduck/tsduck #544: Remove from the PMT the not existing PIDs

stoyanovgeorge, 2020-02-27:
I am having an MPTS stream, which I am trying to modify with TSDuck. The problem is that there are a bunch of ECM and EMM which are referenced in the PMT and when I remove their PIDs, I still see their reference in the PMT table, they are just with 0bps speed.  This is my TSP command: 
```
tsp -P svremove 5400 \
      -P svremove 5401 \
      -P svremove 5403 \
      -P svremove 5404 \
      -P svremove 5411 \
      -P svremove 5412 \
      -P svremove 5421 \
      -P svremove 5422 \ 
      -P filter --pid 0-1200 \
      -P regulate \
      -I ip 232.0.55.4:4000 \
      -O ip localhost:3333 1>/dev/null
```

And this is my Service analysis table: 
```
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 2,372, PID's: 6 (clear: 6, scrambled: 0)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                              Access          Bitrate  |
|   Total  Global PID's .................................                C      132,483 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      132,483 b/s  |
|  0x0000  PAT ..........................................                    C       15,080 b/s  |
|  0x0001  CAT ..........................................                    C       16,700 b/s  |
|  0x0010  DVB-NIT ......................................                C       23,793 b/s  |
|  0x0011  SDT/BAT ......................................                 C        1,675 b/s  |
|  0x0012  EIT ..........................................                      C       74,228 b/s  |
|  0x0014  TDT/TOT ......................................                C        1,005 b/s  |
|  0x1003  Nagravision EMM ..............................        C            0 b/s     |
|  0x1005  Nagravision EMM ..............................        C            0 b/s     |
|  0x1007  Nagravision EMM ..............................        C            0 b/s     |
|  0x1008  NDS EMM ......................................              C            0 b/s     |
|  0x1013  Nagravision EMM ..............................         C            0 b/s    |
|  0x1016  Irdeto EMM ...................................              C            0 b/s     |
|  0x1019  Irdeto EMM ...................................              C            0 b/s     |
|  0x1032  Viaccess EMM .................................            C            0 b/s     |
|=============================================================================|
|  Service: 0x151A (5402), TS: 0x0455 (1109), Original Netw: 0x0001 (1)       |
|  Service name: DMAX HD, provider: BetaDigital                               |
|  Service type: 0x19 (Advanced codec HD digital television service)          |
|  TS packets: 263,034, PID's: 16 (clear: 16, scrambled: 0)                   |
|  PMT PID: 0x0062 (98), PCR PID: 0x02FF (767)                                |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Advanced codec HD digital television service . C   14,691,268 b/s  |
|  0x0022  Teletext (deu, Initial Teletext page) ........ C      251,171 b/s  |
|  0x0032  SCTE 35 Splice Info .......................... C            0 b/s  |
|  0x0062  PMT .......................................... C       15,080 b/s  |
|  0x02FF  AVC video (1440x1080, high profile, level 4.0  C   13,983,776 b/s  |
|  0x0303  AC-3 Audio (deu, clean effects, AC-3, 3/2 (L,  C      439,508 b/s  |
|  0x0305  MPEG-2 Private sections (AIT) ................ C        1,731 b/s  |
|  0x188E  NDS ECM ...................................... C            0 b/s  |
|  0x198E  Nagravision ECM .............................. C            0 b/s  |
|  0x198F  unknown (0x4B64) ECM ......................... C            0 b/s  |
|  0x1A8E  NDS ECM ...................................... C            0 b/s  |
|  0x1B8E  NDS ECM ...................................... C            0 b/s  |
|  0x1B8F  Irdeto ECM ................................... C            0 b/s  |
|  0x1C8E  Nagravision ECM .............................. C            0 b/s  |
|  0x1D8E  Nagravision ECM .............................. C            0 b/s  |
|  0x1D90  Nagravision ECM .............................. C            0 b/s  |
|  0x1E8E  Nagravision ECM .............................. C            0 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
```
I am mostly interested if there is a way to remove the missing PIDs from the PMT without dropping it and then re-inserting the PMT with the corrected PIDs. 

And as a side question is there a possibility to group the services to remove in a list or process only a single service. I also want to remove the CAT table from the stream.

lelegard, 2020-02-27:
Hi,

A PMT can be modified on the fly using the plugin `pmt`. You may remove or add PID references, among other tricks.

For other more sophisticated modifications of the PMT or any other table which cannot be done on the fly using a plugin, the ultimate solution is to extract the input table as an XML file, edit the XML file and reinject the modified table.

Removing the CAT is simply removing PID 1. Then, the easiest way to remove EMM PID's without knowing them is to use the plugin `rmorphan` which removes orphan (unreferenced) PID's. If you have removed the CAT, the EMM PID's become unreferenced.

One last note about your command: You use the plugin `regulate` after removing many services and PID's. The bitrate of the regulation will be incorrect (unless the input device is self-regulated in which case the plugin `regulate` is unnecessary anyway) because the TS bitrate has dropped between the input and the plugin `regulate`, because of the removed PID's. But the plugin `regulate` is not aware of that. So, you should add a plugin `pcrbitrate` just before `regulate` in order to recompute the actual bitrate after PID removal.

Please read the TSDuck user's guide, the reference chapter of `tsp` for an explanation of bitrate handling.

stoyanovgeorge, 2020-02-27:
Thanks Thierry for the reply and the hints, is there a way to remove a couple of services at the same time from the MPTS without passing every time `-P svremove 0xXXXX`?

lelegard, 2020-02-27:
>  is there a way to remove a couple of services at the same time from the MPTS without passing every time -P svremove 0xXXXX?

No, the idea of TSDuck is to have a lot of very simple plugins which do one elementary thing and can be combined in an infinite number of ways (this is the traditional Unix approach). So, the `svremove` plugin only removes one service but you can use any number of them.

stoyanovgeorge, 2020-02-27:
I have changed my command to include the tips you gave me and now the EMM PIDs are not referenced anymore but I still see the ECM PIDs in the PMT table, even though I have excluded them: 
My new command is: 
```
tsp -P svremove 5400 \
      -P svremove 5401 \
      -P svremove 5403 \
      -P svremove 5404 \
      -P svremove 5411 \
      -P svremove 5412 \
      -P svremove 5421 \
      -P svremove 5422 \ 
      -P filter -n -p 1 \
      -P filter -n -p 0x0310-0x1FFC\
      -P pmt -r 0x0310-0x1FFC \
      -P rmorphan \
      -I ip 232.0.55.4:4000 \
      -O ip localhost:3333 1>/dev/null
```
And this is my Service list table: 
```
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 39,548, PID's: 6 (clear: 6, scrambled: 0)                      |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C    2,171,223 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      113,919 b/s  |
|  0x0000  PAT .......................................... C       14,823 b/s  |
|  0x0010  DVB-NIT ...................................... C       23,387 b/s  |
|  0x0011  SDT/BAT ...................................... C        1,647 b/s  |
|  0x0012  EIT .......................................... C       73,073 b/s  |
|  0x0014  TDT/TOT ...................................... C          988 b/s  |
|  0x1FFF  Stuffing ..................................... C    2,057,304 b/s  |
|=============================================================================|
|  Service: 0x151A (5402), TS: 0x0455 (1109), Original Netw: 0x0001 (1)       |
|  Service name: DMAX HD, provider: BetaDigital                               |
|  Service type: 0x19 (Advanced codec HD digital television service)          |
|  TS packets: 196,964, PID's: 16 (clear: 16, scrambled: 0)                   |
|  PMT PID: 0x0062 (98), PCR PID: 0x02FF (767)                                |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Advanced codec HD digital television service . C   10,813,515 b/s  |
|  0x0022  Teletext (deu, Initial Teletext page) ........ C      246,779 b/s  |
|  0x0032  SCTE 35 Splice Info .......................... C            0 b/s  |
|  0x0062  PMT .......................................... C       14,823 b/s  |
|  0x02FF  AVC video (1440x1080, high profile, level 4.0  C   10,118,359 b/s  |
|  0x0303  AC-3 Audio (deu, clean effects, AC-3, 3/2 (L,  C      431,905 b/s  |
|  0x0305  MPEG-2 Private sections (AIT) ................ C        1,647 b/s  |
|  0x188E  NDS ECM ...................................... C            0 b/s  |
|  0x198E  Nagravision ECM .............................. C            0 b/s  |
|  0x198F  unknown (0x4B64) ECM ......................... C            0 b/s  |
|  0x1A8E  NDS ECM ...................................... C            0 b/s  |
|  0x1B8E  NDS ECM ...................................... C            0 b/s  |
|  0x1B8F  Irdeto ECM ................................... C            0 b/s  |
|  0x1C8E  Nagravision ECM .............................. C            0 b/s  |
|  0x1D8E  Nagravision ECM .............................. C            0 b/s  |
|  0x1D90  Nagravision ECM .............................. C            0 b/s  |
|  0x1E8E  Nagravision ECM .............................. C            0 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
```
So either I am doing something completely wrong or `-P pmt -r 0x0310-0x1FFC` doesn't seem to work.

lelegard, 2020-02-27:
>  -P pmt -r 0x0310-0x1FFC doesn't seem to work.

In a PMT, ECM PID's are not referenced as components of the service. They are declared in CA descriptors. To remove reference to all ECM streams, you need to remove all CA descriptors: `-P pmt --remove-descriptor 9` (where 9 is the tag for CA descriptors).

>  the output doesn't accept -O ip 3333 notation and complains that this is not a multicast address, but the -I ip 3333 is absolutely fine.

This is logical. On input, you indicate a _local_ port on which you listen. The multicast address is optional because you can listen to anything on a local port, including unicast traffic. Specifying a multicast address is just here to join a multicast group if you need to.

On output, you must specify where you want to send the packet. So, you need to specify a _remote_ address (multicast or unicast) and port. By specifying `-O ip 3333`, where do you expect your packets to go?

stoyanovgeorge, 2020-02-27:
thanks for the hints once again, Thierry, really helpful. Specifying `-O ip 3333` I am expecting my packages to be directed to `127.0.0.1:3333` and even defining this as an address didn't help, so I needed to define `localhost:3333` instead.

lelegard, 2020-02-27:
> Specifying -O ip 3333 I am expecting my packages to be directed to 127.0.0.1:3333

No, there is no reason for this. Why should it be like this?

> even defining this as an address didn't help, so I needed to define localhost:3333

The command `tsp ... -O ip 127.0.0.1:3333` is accepted. Just tested, it works.

>  the -O ip directive expects a multicast as an input.

Not at all. You can send on any address, unicast or multicast.

Except some network management rules such as IGMP, unicast and multicast are identical. An IP datagram is just an IP datagram. The fact that the destination address is considered as multicast is just a convention, a conventional range of addresses.

The output plugin makes no difference between multicast and unicast.

The input plugin is slightly different because, in the case of multicast, the client must send an IGMP join request to ask the routers to route the multicast packets. This is the only reason why the multicast address needs to be specified on the input plugin. But, except this initial join request, the rest of the input plugin is identical for multicast and unicast. It can receive both, indifferently.

stoyanovgeorge, 2020-02-28:
Yes, you are right about `-O ip 127.0.0.1:3333` for some reason it wasn't working for me yesterday, but I have updated at some point the version so maybe it is because of that. Anyway, thanks a lot for your help, Thierry you have done a terrific job with TSDuck!

lelegard, 2020-02-28:
Remember that, as explained [here](https://tsduck.io/doxy/contributing.html), TSDuck is made by TV engineers for TV engineers, not for hackers. Hackers are banned. Publishing the structure of a transport stream where a Nagra/NDS/Irdeto-protected service has clear audio and video is not very clever. Misunderstandings could happen and you could be banned...

stoyanovgeorge, 2020-03-01:
I am not doing anything illegal. As stated before, I am just trying to remove some PIDs from the PMT and actually with your help it worked. I am not trying to hack the service, just trying to create a clean PMT, that's all.

closed 2020-03-01 by stoyanovgeorge