tsduck/tsduck #546: Question : How to inject dvb subtitles on the fly

antoineagscom, 2020-03-03:
Hello,

I'd like to inject subtitles into a particular pmt service, however i only identify tags in pmt xml file by example : 

[image]

I can extract subtitles with .srt format, but plugin inject accept only xml or binary files !

How to inject subtitles content on the fly ?
Can i use plugin "mux" or "inject" ?

lelegard, 2020-03-03:
Subtitles are PES data which must be synchronously muxed with the video and audio. So, you need a MUX. And TSDuck is not a MUX. FFmpeg is more appropriate to mux audio, video, subtitles.

Additionally, if you need to convert your SRT file into DVB subtitles or Teletext, you need a dedicated tool for that.

antoineagscom, 2020-03-04:
Thank's for your fast reply,

I'm trying to display an alert on the user's screen like pop-up message, that's why I wanted to go through the subtitles.

I see a plugin "spliceinject" for use SCTE35 but the problem of this method (if i understand), SCTE35 replace video and/or audio PID of PMT identify by service_id, and stop the current stream.

Do you know a solution or a system on dvb stream to have an alert on the fly ?