tsduck/tsduck #547: TsSwitch as IP stream failover

mks75, 2020-03-04:
Hi, I want to perform fail-over between two MPEG-TS multicast IP streams. Can I do so with TsSwitch? Please share the commands as I am new to this.

Thanks
MKS

cbernander, 2020-03-04:
To perform automatic (and continuous) fail-overs between inputs 239.1.1.1:5000 and 239.2.2.2:5000 on an output 239.100.100.100:5000 using tsswitch, the syntax should go something like this:

`tsswitch --infinite --receive-timeout 1000 -I ip 239.1.1.1:5000 -I ip 239.2.2.2:5000 -O ip 239.100.100.100:5000`

This example should output the first input stream until no data is received on the first input stream for 1000 ms, in which case it will output the second input stream instead until no data is received on the second input stream for 1000 ms, in which case it will output the first input stream again and so on.

It's been a while since I tried this myself so I might have gotten something wrong but go ahead and give it a try.

Please consult the manual regarding the usage of the options specified in this particular example, as well as other available options such as _--fast-switch_ and _--primary-input_.

mks75, 2020-03-04:
Thanks @cbernander ,  I can see the output. 
I have one obervation. The output multicast should have packet size of 1316, but in the output I am seeing at times this value is less than 1316.
is there a parameter to include here that I am missing.

17:42:05.761960 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
17:42:05.772192 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
17:42:05.772196 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
17:42:05.772199 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
**17:42:05.773738 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 940
17:42:05.773742 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 376**
17:42:05.783802 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
17:42:05.783806 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316
17:42:05.783809 IP 10.20.20.18.48230 > 239.4.4.4.1234: UDP, length 1316

please advise.
Thanks
MKS

cbernander, 2020-03-04:
Try adding the options _--enforce-burst_ and _--packet-burst_ to the ip output, e.g.:

`... -O ip --packet-burst 7 --enforce-burst 239.4.4.4:1234`

mks75, 2020-03-04:
Thanks a lot @cbernander , the output now has consistent packet size of 1316.

Thanks
MKS

closed 2020-03-04 by mks75