tsduck/tsduck #548: Semi-BUG: incorrect time in verbose mode of plugin timeref

lars18th, 2020-03-04:
Hi @lelegard ,

I detected this _small_ error when using the `timeref` plugin in Windows. See this example:
```
tsp.exe -v -b 23418685 -I file -i ./test.ts -P timeref --start system --eit -O hides
-f 482000000 -g "1/16" -h "2/3"
```

The idea is to replace the time in the output MPTS with the system time. Well, it works right, but the verbose output indicates:
```
* timeref: current system clock is 1601/01/01 00:00:00.000
* file: initial input bitrate is 23,418,685 b/s
* hides: using device 0: "IT9507 TX Filter" with nominal output bitrate of 0 bits/s
* timeref: adding 7,438,718,190 milliseconds to all event start time in EIT's
```

I'm sure the current system clock is not in this date. Even the output has correct time, the debug I feel is printing using incorrect format.

Only for your information.
Regards.

lelegard, 2020-03-04:
1601 is Windows Epoch. Such  a date is the symptom of an uninitialized date, a zero value. I will check.

lelegard, 2020-03-04:
OK, it's a bug. See that code:
~~~
bool ts::TimeRefPlugin::getOptions()
{
    ...
        if (start == u"system") {
            _startref = Time::CurrentUTC();
            tsp->verbose(u"current system clock is %s", {ts::UString(_timeref)});
        }
~~~
Of course, the display should use `_startref`, not `_timeref` which is initially zero, so in year 1601 on Windows. Will be fixed.

lelegard, 2020-03-16:
Fixed

closed 2020-03-16 by lelegard