tsduck/tsduck #549: `TS_FACTORY_REGISTER` macro seems not to work correctly with static-link

masnagam, 2020-03-07:
Hi @lelegard,

I found an issue in `v3.19-1520`, which occurs only when linking statically.

I haven't checked whether the same issue occurs in the master branch.
However, at least, definitions of macros related to this issue have not been changed.

## Reproduction Steps

Add the following code somewhere in a source file of an executable:

```c++
ts::UStringList names;
ts::TablesFactory::Instance()->getRegisteredDescriptorNames(names);
for (const auto& name : names) {
  std::cerr << name << std::endl;
}
```

And then **static**-link the executable against TSDuck.

## Expected Behavior

All descriptors should be shown.

## Actual Behavior

Only several descriptors are shown.

All descriptors are shown when **dynamic**-linking against TSDuck.

## Possible Root Cause

Probably, unused static variables are removed due to the compiler optimization.

lelegard, 2020-03-07:
With static link, a module from the static library is included in the final application only if it is referenced. Since tables and descriptors are self-registering modules, most of them are not referenced and consequently not included in the link.

To be sure to collect all tables and descriptors in a static application, you need to create an instance of the class `StaticReferencesDVB`. This is a dummy class which forces a reference to all tables and descriptors. See the details [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/static/tsStaticReferencesDVB.h#L40) or [here](https://tsduck.io/doxy/classts_1_1_static_references_d_v_b.html).

masnagam, 2020-03-07:
Thank you for your quick answer.
OK.  I'll apply similar solution in my project.

masnagam, 2020-03-07:
Sorry, I have one more question.

Do you have a plan to create a similar class to hold the references to `ts::TablesFactory::Register` static variables?

`ts::TablesFactory::Register` for each descriptor class is defined a file-scope static variable.  Therefore, it cannot be accessed from outside the library.

masnagam, 2020-03-07:
I found a workaround for this issue.

We can register any descriptor class by using code like below:

```c++
// in application source
#define MY_XML_NAME u"stream_identifier_descriptor"
#define MY_DID ts::DID_STREAM_ID
#define MY_STD ts::STD_DVB
TS_XML_DESCRIPTOR_FACTORY(ts::StreamIdentifierDescriptor, MY_XML_NAME);
TS_ID_DESCRIPTOR_FACTORY(ts::StreamIdentifierDescriptor, ts::EDID::Standard(MY_DID));
TS_FACTORY_REGISTER(ts::StreamIdentifierDescriptor::DisplayDescriptor, ts::EDID::Standard(MY_DID));
```

closed 2020-03-08 by masnagam